hku-mars/Point-LIO #1: What does "High-Bandwidth" mean?

jiejie567, 2023-02-02:
Hi @Joanna-HE , I can't wait to see your accepted paper，your work in the video really impresses me deeply. I have a question about the meaning of  "High-Bandwidth".

jiejie567, 2023-02-03:
What's more，in the high vibration situation, what makes Point-Lio more accurate than Fast-Lio?

Joanna-HE, 2023-02-03:
Hi thanks for your interest in our work. 

1. We define odometry bandwidth in analogy to the bandwidth of a dynamic system, which is the frequency where the system gain drops below 0.707. An odometry bandwidth represents how fast motion an odometry can estimate.

2. The high vibration would result in severe motion distortion in one scan of LiDAR points. Point-LIO is distortion free, that makes it more accurate than Fast-LIO.

jiejie567, 2023-02-03:
Thanks, @Joanna-HE . I have known about it. Best wishes!

closed 2023-02-03 by jiejie567