hku-mars/Point-LIO #2: State of the project

EloyAldao, 2023-03-01:
Dear developer

Today you deleted the YouTube video about Point-LIO

Does this means that the project will not be public released or it means that there will be a new video because the paper was accepted and the code will be available?

Hoping the best

Thanks in advance

Joanna-HE, 2023-03-08:
Thanks for your interest in our work! The YouTube video is deleted due to a policy of the Science Robotics. After the paper of Science Robotics is published on 15th March, 2023, the video would be released again.

Best regards,

Dongjiao He

Joanna-HE, 2023-03-08:
Hi, thanks for your interest in our work! The video is deleted for a policy of a journal. The video will be released again after 15th March, 2023, and the code would also be public released.

Best regards

Dongjiao He

closed 2023-05-19 by Joanna-HE