hku-mars/Point-LIO #8: mid-70, mid-40 or other livox serial LiDAR

MasterBicycle, 2023-03-25:
Hello author, I tested your algorithm works well on Velodyne and Ouster data, but the following problem occurs when running on livox laser. I tried many ways, but didn't find a solution, can you give me some advice?
[image]

MasterBicycle, 2023-03-25:
[image]

Joanna-HE, 2023-03-26:
请问你用的是livox的内置IMU嘛？检查一下yaml文件中，是否将IMU的量程"acc_satu"和“gyro_satu”以及加速度计的单位“acc_norm”设置对了呢？

MasterBicycle, 2023-03-26:
感谢你的及时回复，是内置的IMU，这些值采用的是文件中默认值
[image]

MasterY0416, 2023-03-26:
哦。应该是我加的代码有关系，将点云地图和重力系方向对齐。您有一些好的建议吗
[image]

MasterBicycle, 2023-03-26:
哦，应该是和我加的代码有关系，将第一帧点云和重力系方向对齐。但是在Velodyne和Ouster数据下都可以正常运行，您有好的建议吗
[image]

MasterBicycle, 2023-03-26:
谢谢作者的及时回复，现在代码能正常运行了。不过还有个疑问，当把第一帧点云通过重力系对齐后，'use_imu_as_input'设置成1，相对来说对状态估计的结果有很大的影响，我还在进一步学习您的代码，感谢和您的交流。
[image]

Joanna-HE, 2023-03-26:
Point-lio是将第一帧的IMU坐标系设为world系的，所以如果要做第一帧点云通过重力系对齐的话，相当于修改了world系，需要相应修改当前状态量中的IMU相对与world系的旋转。

MasterY0416, 2023-03-27:
恍然大明白，“花”

MasterBicycle, 2023-03-27:
恍然大明白，“花”

closed 2023-03-27 by MasterBicycle