hku-mars/Point-LIO #9: Add save trajectroy function

Taeyoung96, 2023-03-26:
@Joanna-HE   
Hi, Thanks for your great work!  
I've made a PR that adds a function to store the trajectory function at #5.  

I can't hear any comments on this PR, so I'm opening an issue.   
I would be happy to merge the PR if it would help. 😄   

If there are any additional tests that need to be done or guidelines that need to be followed in order for the PR to merge, I'll do it.   

I'd be grateful for your response.  

Thanks,

Joanna-HE, 2023-03-26:
Hi TaeYoung Kim,

Thanks a lot for your work! I have noticed your pull request, but still need some time to work on your request. I will contact you as soon as possible after there is an update of this request.

Best regards,

Dongjiao He
________________________________
寄件者: TaeYoung Kim ***@***.***>
寄件日期: 2023年3月26日 下午 07:33
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Mention ***@***.***>
主旨: [hku-mars/Point-LIO] Add save trajectroy function (Issue #9)

@Joanna-HE<https://github.com/Joanna-HE>
Hi, Thanks for your great work!
I've made a PR that adds a function to store the trajectory function at #5<https://github.com/hku-mars/Point-LIO/pull/5>.

I can't hear any comments on this PR, so I'm opening an issue.
I would be happy to merge the PR if it would help. 😄

If there are any additional tests that need to be done or guidelines that need to be followed in order for the PR to merge, I'll do it.

I'd be grateful for your response.

Thanks,

Taeyoung96, 2023-03-26:
@Joanna-HE   
Thank you for your quick response!  
If there's anything else I should do, please get back to me.

closed 2023-03-26 by Taeyoung96