hku-mars/Point-LIO #10: IMU initializing

xbnz-1997, 2023-03-27:
Hi，author! Thanks for your great work.When I run your code with my datasets, the following content appears. It remained at 67% until the end of the dataset play, accompanied by a big drift.

[image]

[image]

Joanna-HE, 2023-03-27:
There is a unclear indication in the current code of Point-LIO, when the IMU initialization finished, the 100% would not show in the terminal screen. I have made this indiction of IMU initialization process clear in the new commit. and the drift in your pic has nothing to do with the IMU initialization process, which type of LiDAR do you use? And have you made a right value for 'satu_acc', 'satu_gyro' and 'acc_norm' according to the IMU you use?

xbnz-1997, 2023-03-28:
The type of lidar is livox avia, same as you, do I still need to make changes to it?

Joanna-HE, 2023-03-28:
Yes, should make changes according to the measuring ranges and units of the IMU you use.
________________________________
寄件者: xbnz-1997 ***@***.***>
寄件日期: 2023年3月28日 上午 09:55
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] IMU initializing (Issue #10)

The type of lidar is livox avia, same as you, do I still need to make changes to it?

xbnz-1997, 2023-03-28:
The IMU what I use is livox avia lidar build-in，so could you tell me how to change the value for 'satu_acc', 'satu_gyro' and 'acc_norm'?

xbnz-1997, 2023-03-28:
By the way, when will the datasets be published?

Joanna-HE, 2023-03-31:
Firtstly, make sure that you know the measuring ranges of the IMU you used (3 g and 35 rad/s for built-in IMU of livox avia by default, but they could be changed according to your implementation), and the unit of the accelerator of your IMU (g by default, but could also be changed). If all the values are default without any changes, satu_acc = 3, satu_gyro = 35, and acc_norm = 1.0. If you make any changes, the exact values for those parameters could be modified according to the comments in yaml file.

The datasets are already published.

Thanks for your patience!

xbnz-1997, 2023-04-01:
ok, thank you very much!

closed 2023-04-01 by xbnz-1997