hku-mars/Point-LIO #11: h_model_input 代码逻辑问题

cxyeSuper, 2023-03-28:
为了表述清晰和节省时间，很抱歉直接使用中文。

前要：论文中提出对点云帧中每个点云分别按获取时间节点进行卡尔曼的更新，因此无需对整个点云包进行点云畸变处理。

但代码中的实现我认为可能存在问题，如果是我理解有误请指正：

h_model_input 函数中for (int j = 0; j < time_seq[k]; j++){...}的代码逻辑为：
从每个点云帧中按点云时间顺序抽取点云进行world近邻搜寻，但其中有一些问题:
t1: 针对点云第一个点进行最近邻搜索后，更新状态信息。
t2: 针对点云第一个点和第二个点进行最近邻搜索，更新状态信息。
...
tn：针对点云包中第一个点至第n个点进行最近邻搜索，更新状态信息。
但是，在每次状态更新时，如tn时间点，第1个点至第n-1个点都是直接用当前状态进行的body坐标系到world坐标系的变换，这实际上引入了点云畸变。

而在逻辑上，在tn时刻中，第一个点至第n-1个点在世界坐标系中的位置理应已经固定。
因此，我认为代码逻辑应该改为：在每一个时刻更新后，即刻固定所处理点云在世界坐标系中的位置。

Joanna-HE, 2023-03-28:
代码就是在每一个时刻更新后，即刻固定处理点云在世界系的位置的。

cxyeSuper, 2023-03-28:
您回答以后刚仔细看了下代码，是我先入为主没看time_compressing函数，错误理解time_seq和idx的含义了。
实际上您的代码实现和我上述描述的滤波优化逻辑不同，才有了上述本质就是错误的疑问，很抱歉。
您的代码实现确实和论文中的伪代码一致。