hku-mars/Point-LIO #15: how to deal with my lidar whose point has negative sample time

ouguangjun, 2023-04-03:
My LiDAR is velodyne 16 and  all the point sample time are nagative (about -100~0 ms).

Joanna-HE, 2023-04-04:
Hi thanks for your interest in our work, but we did not encounter the negative timestamps when using the velodyne 16, maybe you could provide more information about the negative sample timestamps, and we can talk about here.

Joanna-HE, 2023-04-04:
Hi, I have an assumption that has not been testified. The negative timestamps indicate that your velodyne does not provide the timestamp for each point. At this time, you should give a timestamp for each point based on the message header time and the scanning pattern of the velodyne.
________________________________
寄件者: Ou guangjun ***@***.***>
寄件日期: 2023年4月4日 下午 03:16
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] how to deal with my lidar whose point has negative sample time (Issue #15)

I print the "pointcloud->points[i].time" and all of them are nagative (-0.1s~0s). The frequence of LiDAR is 10 hz. May be I should ask for the producer. Thanks

Joanna-HE, 2023-04-04:
You can try the Point-LIO, maybe it is able to handle.

ouguangjun, 2023-04-04:
I have tried the point-lio before  and found that the code will measure the sample time for each point when the timestamp is not provided. Thank u for your reply!

closed 2023-04-04 by ouguangjun