hku-mars/Point-LIO #18: 保存PCD文件报错

78748547, 2023-04-06:
你好，我修改了**mapping_velody16.launch**文件中的**pcd_save_en**的参数为true。当我成功建图并结束终端后，终端提示以下图中的报错信息，Point-LIO文件夹中也没有生成PCD文件夹，想请问一下这种问题如何解决，谢谢。
[image]

Gatsby23, 2023-04-06:
> 你好，我修改了**mapping_velody16.launch**文件中的**pcd_save_en**的参数为true。当我成功建图并结束终端后，终端提示以下图中的报错信息，Point-LIO文件夹中也没有生成PCD文件夹，想请问一下这种问题如何解决，谢谢。 [image]

Open这里出问题大概率是没有权限或路径出错

FPSychotic, 2023-04-06:
Just create a folder called PCD on point-lio folder into SRC

78748547, 2023-04-06:
> Just create a folder called PCD on point-lio folder into SRC

Thanks for your reply. It is very useful.

closed 2023-04-06 by 78748547

wyz1028, 2024-02-27:
请问保存点云这个操作的代码应该加到launch文件的哪一行？

Joanna-HE, 2024-03-01:
在.yaml文件中将pcd_save_en的值改为true即可。
________________________________
寄件者: wyz1028 ***@***.***>
寄件日期: 2024年2月27日 下午 09:58
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 保存PCD文件报错 (Issue #18)

请问保存点云这个操作的代码应该加到launch文件的哪一行？