hku-mars/Point-LIO #19: Only registered cloud is published livox mid-360

FPSychotic, 2023-04-06:
hi, I have point-lio odometry working without any fine tunning, but it looks working very good. 
My problem is I cannot get a dense pointcloud as only  registered cloud is being published. 

thanks
[image]
[image]

Joanna-HE, 2023-04-07:
Thanks for your interest in our work! For your information, the parameters are not needed to be fine tuned, excepting those are related to IMU, since the point-lio is designed to be robust to IMU saturation, those parameters should be modified according to the IMU you use. 

And there is not a dense map produced by the Point-LIO yet, since the Point-LIO registers a lidar point into the map after it calculates the robot localization at the exact sampling time of this lidar point. For those points that are not processed, the accurate localization is not obtained. Maybe a function for inaccurate dense map could be added into the Point-LIO in the near future.

FPSychotic, 2023-04-07:
Ohh, I see, in my opinion is totally normal there is not dense pointcloud as it is a odometry package, the only thing that made me think in that was a YouTube video and some pictures,where looked dense at my eyes, but maybe it was other package with Point-lio as odometry root.
It is very light for CPU, you did a great job , thanks by share it.

FPSychotic, 2023-04-07:
Ohh, I see, in my opinion is totally normal there is not dense pointcloud as it is a odometry package, the only thing that made me think in that was a YouTube video and some pictures,where looked dense at my eyes, but maybe it was other package with Point-lio as odometry root.
It is very light for CPU, you did a great job , so smart

closed 2023-04-07 by FPSychotic

FPSychotic, 2023-04-08:
ohh, discovered going outdoors it will show the map just like fast-lio does, that kind of map is what I was looking for, It is what I understand by dense, but maybe technically is other thing.