hku-mars/Point-LIO #20: 跑数据集的时候总是飘

zilong-ding, 2023-04-07:
[image]
[image]
[image]

我在自己电脑上跑onedrive里面的数据集，分别使用
roslaunch point_lio mapping_avia.launch和
roslaunch point_lio mapping_horizon.launch
对四个数据集（PULSAR.bag   pendulum.bag   racing_drone.bag  'spinning_platform (1).bag'）进行测试，
但是总是有比较明显的偏移（一般是前期比较稳，后面会飘），是相关参数设置不对吗？

Joanna-HE, 2023-04-07:
pulsar的imu我们重新刷了一下，陀螺仪的量程是17.5，所以要把avia.yaml的’satu_gyro'改为17.5

Joanna-HE, 2023-04-07:
文章里有写，我刚刚在readme里也添加了这个说明

zilong-ding, 2023-04-07:
[image]
谢谢，效果很棒

closed 2023-04-08 by zilong-ding