hku-mars/Point-LIO #26: last commit 7701f1c brakes Point-Lio in arm64 cpu

FPSychotic, 2023-04-22:
Hi I installed the last version with conmmit [7701f1c] in two machines x86, worked good. I installed in two arm64 machines and in two different ways , modifying Point-lio to use Livox_ros_driver2:custom_msg and using the livox_ros_driver, in any case, with arm64 (jetson Orin NX 16gb) always crash in some point losing odometry and the map gets crazy jumping.

Could you please, at least make a different branch with the point-lio version previous to that commit?

Thanks.

Joanna-HE, 2023-04-24:
And for your information, in the commit [7801f1c], I added the normalized() function of Eigen for the quaternion calculations. I am not sure if the problems is caused by the quaternion calculations of Eigen.

FPSychotic, 2023-04-24:
It works before that commit, so I understand if it is the only change you made , there is some issue with it, maybe a bug in the ARM64 version of Eigen or maybe the ARM64 the version installed is different than x86.

Joanna-HE, 2023-04-24:
I delete normailized() of Eigen in the branch 'test'. I would be grateful if you could help to check it on ARM64.

FPSychotic, 2023-04-24:
Yes , I will . I will give more detailed info of PC,Eigen versions and kernel in case is useful

FPSychotic, 2023-04-24:
@Joanna-HE Hi, I did the tests and installed my stable version, the test branch (should be the same) and the main branch. The main branch brakes in two of 3 bags always in the same points. I did the same, install the 3 versions in a X86 and worked in all of them.
All of them with Point-Lio aiming to livox_ros_driver2
No sure if this issue is only in that device, Jetson Orin NX, I would format it and make a clean installation to check but Nvidia Jetsons are a mesh to reinstall, the setup is , jetpak 5.1.1, Eigen 3.3.7-2, PCL 1.10, GTSAM 4.0.
I hope this help, in my opinion would be nice keep the branch in case other person have the same issue, also I didn't have have instability or randomness in this device.

closed 2023-04-24 by FPSychotic