hku-mars/Point-LIO #33: Last commit fails to map with bags.

FPSychotic, 2023-05-20:
hi , I'm using ros driver 2 and where before the bags worked good, now the bags from point-lio or from fast-lio fails as the odometry gets crazy, like when the imu is bad integrated or out synchro. 
In real time it looks works good, just detected the issue in the bags.
Same bags with ring field in the topinc of the point cloud , as velodyne, will work.
Just for your info, for me it is not a problem as I can use ring information and will work or use the main branch, but this info maybe help you to improve.

cheers!

Joanna-HE, 2023-05-22:
Thanks for your information and tests! Time stamp of LiDAR points is needed for the Point-LIO, and the preprocess of the code is supposed to get the time stamp from a specific data structure by hard coding. I will close this commit until there is a same problem encountered by others. And I will try to improve the Point-LIO based on this information.

Thanks!

closed 2023-05-22 by Joanna-HE

Joanna-HE, 2023-05-22:
I have thought of a problem that may result in a crash in rosbag but work well in real time. When you run Point-LIO by playing rosbag, what is the play speed of the rosbag? Playing the rosbag at its normal speed may be possible to handle the crash problem. This problem is a side effect of ikd-tree that rebuilds the map in the second thread than the main thread.