hku-mars/Point-LIO #37: use_imu_as_input参数的设置

WeilongXia, 2023-07-03:
作者你好，非常感谢能分享这么棒的工作！但我在实际测试的时候发现了一些问题，use_imu_as_input设置成true比设置成false效果更好，这是否不太合理呢？这是我的测试数据集：https://1drv.ms/u/s!AjkIEg1kWUNeiAvyTCBBnR-jxLE1?e=jXvtla（使用的传感器是Livox MID-360，并且是倒置安装的）
以下是我设置的参数：
```
common:
    lid_topic:  "/livox/lidar" 
    imu_topic:  "/livox/imu" 
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency
    time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 1 
    scan_line: 6
    timestamp_unit: 1           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1.0 

mapping:
    imu_en: true
    start_in_aggressive_motion: false # if true, a preknown gravity should be provided in following gravity_init
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.005 # = 1 / frequency of IMU
    satu_acc: 3.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 1.0 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.001 # 0.001; 0.01
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.0001 
    b_gyr_cov: 0.0001 
    imu_meas_acc_cov: 0.1 #0.1 # 0.1
    imu_meas_omg_cov: 0.1 #0.01 # 0.1
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 100.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.806] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.806] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [ 0.011, 0.02329, -0.04412 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ]

odometry: 
    publish_odometry_without_downsample: false
    I_t_B: [ 0.0, 0.0, -0.2 ]
    I_R_B: [ 1, 0, 0,
             0, -1, 0,
             0, 0, -1 ]

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```
期待回复，祝好～

Joanna-HE, 2023-07-05:
你好！感谢你对我们工作的关注！可以展示下两个建图结果，和你觉得哪里有差异嘛？

祝好

WeilongXia, 2023-07-08:
感谢回复！
`use_imu_as_input`设置为1的时候，建图结果及估计轨迹为：
[image]
[image]
和GPS对比，发现轨迹估计是很正常的，建图结果看起来也非常棒。
`use_imu_as_input`设置为0的时候，建图结果及估计轨迹为：
[image]
[image]
可以看出，建图有重影并且轨迹在y轴上一直有漂移。（测试场景是一个大的操场，无人机在操场中央起飞，飞至16米空中再降落）
btw，建图可以看到飞机脚架的“鬼影”如下图所示：
[image]
但我已经将程序中的`blind`参数设置为0.5了，按理说应该去掉了无人机自身的点才对，不知道是不是因为可视化的是未经过处理的点云呢？

Joanna-HE, 2023-07-11:
你好~请问你用是ubuntu20.04及以上嘛？我在unbuntu20.04上用point-lio测试了链接中的数据，一个是在use_imu_as_input=0时，不会有y方向的偏移，一个是没有轨迹旁的鬼影。如果系统是20.04的话，可以试试看拉取最新的commit代码进行测试呢？
________________________________
寄件者: Weilong Xia ***@***.***>
寄件日期: 2023年7月8日 下午 01:36
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

感谢回复！
use_imu_as_input设置为1的时候，建图结果及估计轨迹为：
[Screenshot from 2023-07-08 11-23-39]<https://user-images.githubusercontent.com/61641798/251937532-771c1042-1b2c-4afe-bf58-b5cb3f02effb.png>
[Screenshot from 2023-07-08 11-23-59]<https://user-images.githubusercontent.com/61641798/251937531-3eb29d0a-2d90-4b88-8811-fc7947800507.png>
和GPS对比，发现轨迹估计是很正常的，建图结果看起来也非常棒。
use_imu_as_input设置为0的时候，建图结果及估计轨迹为：
[Screenshot from 2023-07-08 11-29-34]<https://user-images.githubusercontent.com/61641798/251937595-5ed271c7-27c4-4274-b950-21d7669b4390.png>
[Screenshot from 2023-07-08 11-30-02]<https://user-images.githubusercontent.com/61641798/251937605-7fd4aa00-62c3-4309-9938-08a2b21cf975.png>
可以看出，建图有重影并且轨迹在y轴上一直有漂移。（测试场景是一个大的操场，无人机在操场中央起飞，飞至16米空中再降落）
btw，建图可以看到飞机脚架的“鬼影”如下图所示：
[Screenshot from 2023-07-03 16-48-29]<https://user-images.githubusercontent.com/61641798/251937799-bed7f85b-0200-4351-9059-8ec11b37d9b2.png>
但我已经将程序中的blind参数设置为0.5了，按理说应该去掉了无人机自身的点才对，不知道是不是因为可视化的是未经过处理的点云呢？

WeilongXia, 2023-07-12:
您好，我用了最新提交的代码进行了测试，satu_acc设置为4.0，结果是这样的：
[image]
可以看出y轴还是误差比较大
我使用的系统是Ubuntu20.04，PCL版本是1.10
[image]
请问还有其他可能原因导致这种现象吗？

Joanna-HE, 2023-07-12:
试试看将lidar_meas_cov设为0.1，效果应该会有比较明显的改善
________________________________
寄件者: Weilong Xia ***@***.***>
寄件日期: 2023年7月12日 上午 10:04
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

您好，我用了最新提交的代码进行了测试，satu_acc设置为4.0，结果是这样的：
[Screenshot from 2023-07-12 09-58-57]<https://user-images.githubusercontent.com/61641798/252834790-1b772a4b-adcc-4eee-ac9f-7179592581a2.png>
可以看出y轴还是误差比较大
我使用的系统是Ubuntu20.04，PCL版本是1.10
[Screenshot from 2023-07-12 09-59-08]<https://user-images.githubusercontent.com/61641798/252834899-38358a24-aed0-4348-bcd6-3aa1c0c36796.png>
请问还有其他可能原因导致这种现象吗？

Joanna-HE, 2023-07-12:
或者调小imu_meas_acc_cov和imu_meas_omg_cov到0.01。因为距离比较远的雷达测量的误差是会比近距离时大的。
________________________________
寄件者: ***@***.*** ***@***.***>
寄件日期: 2023年7月12日 下午 12:55
收件者: hku-mars/Point-LIO ***@***.***>
主旨: 回覆: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

试试看将lidar_meas_cov设为0.1，效果应该会有比较明显的改善
________________________________
寄件者: Weilong Xia ***@***.***>
寄件日期: 2023年7月12日 上午 10:04
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

您好，我用了最新提交的代码进行了测试，satu_acc设置为4.0，结果是这样的：
[Screenshot from 2023-07-12 09-58-57]<https://user-images.githubusercontent.com/61641798/252834790-1b772a4b-adcc-4eee-ac9f-7179592581a2.png>
可以看出y轴还是误差比较大
我使用的系统是Ubuntu20.04，PCL版本是1.10
[Screenshot from 2023-07-12 09-59-08]<https://user-images.githubusercontent.com/61641798/252834899-38358a24-aed0-4348-bcd6-3aa1c0c36796.png>
请问还有其他可能原因导致这种现象吗？

WeilongXia, 2023-07-12:
好的！我已经尝试过修改lidar_meas_cov，发现效果有了很大的提升：
[image]
但是使用同一数据连续测试两次发现，在资源受限较高的情况下（比如第一次测试我开了其他程序导致CPU占用率比较高），lio的效果会有明显的下降（定位突变了一次），想问下程序里哪块会有随机性吗？
还想请教一下，在特别开旷的场景下（比如机场，Lidar可能只能扫到地面），point-lio作为定位源来使用可靠程度高吗？
我会尝试调整不同的参数来查看对比效果的好坏，非常感谢您的帮助～

Joanna-HE, 2023-07-13:
Point-LIO目前使用的地图是ikdtree，ikdtree是双线程运行且是在等主线程空闲时才会修改tree的结构，这个特性可能会在运行资源吃紧时造成随机性。地图结构可以自行选择换成其他的。

完全的退化场景，Point-LIO不会飞掉，但是也可能会定位不准的。
________________________________
寄件者: Weilong Xia ***@***.***>
寄件日期: 2023年7月12日 下午 01:20
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

好的！我已经尝试过修改lidar_meas_cov，发现效果有了很大的提升：
[Screenshot from 2023-07-12 13-11-56]<https://user-images.githubusercontent.com/61641798/252861532-09f4c228-e09f-4768-9cf5-af434d2c6db2.png>
但是使用同一数据连续测试两次发现，在资源受限较高的情况下（比如第一次测试我开了其他程序导致CPU占用率比较高），lio的效果会有明显的下降（定位突变了一次），想问下程序里哪块会有随机性吗？
还想请教一下，在特别开旷的场景下（比如机场，Lidar可能只能扫到地面），point-lio作为定位源来使用可靠程度高吗？
我会尝试调整不同的参数来查看对比效果的好坏，非常感谢您的帮助～

WeilongXia, 2023-07-13:
好的，非常感谢您的回复，我再研究研究～

closed 2023-07-13 by WeilongXia

giangdao1402, 2023-09-25:
Hey my lidar is set upside down like you, how can you rotate the map , my map is invert

WeilongXia, 2023-09-25:
把yaml文件中的gravity_align参数设置为true

giangdao1402, 2023-09-25:
**I already set gravity_align to true, but it still inverse**	
common:
    lid_topic:  "/ouster/points"
    #imu_topic:  "/ouster/imu"
    #imu_topic: "/imu_data"
    imu_topic: "/mavros/imu/data_raw"
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency 
    time_lag_imu_to_lidar: 1684841499.635005 #1684841524.65910959  #1684841499.635005 #1694165984.83338928 #1694168642.17952871 #0.006376 #-0.48057733 #1684841524.65910959 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 3 # 2 #velodyn	                # 1 Livox Avia LiDAR
    scan_line: 16 # 32 #velodyne 6 avia
    timestamp_unit: 3           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1.0

mapping:
    imu_en: true
    start_in_aggressive_motion: false # if true, a preknown gravity should be provided in following gravity_init
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.001 # = 1 / frequency of IMU
    satu_acc: 30.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 25 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.81 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.01 # 0.01
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.07 
    b_gyr_cov: 0.3 
    imu_meas_acc_cov: 0.1 #0.1 # 2
    imu_meas_omg_cov: 0.1 #0.1 # 2 
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 300.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.810]  #[3.580628, 0.025035, -9.133134] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [0.0, 0.0, 0.0] #[-0.160931, 0.337930, -0.054396] #[-0.277508, -0.179171, -0.014018] #[0.0, 0.0, 0.0]
    extrinsic_R:  [1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ] 
                  #[ 0.033039, -0.822540, 0.567746,
                   #-0.987192, -0.115566, -0.109983,
                    #0.156078 , -0.556841, -0.815824 ]

odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

[image]

hr2894235132, 2024-05-29:
> 好的！我已经尝试过修改lidar_meas_cov，发现效果有了很大的提升： ![Screenshot from 2023-07-12 [...]

你好，想问一下lidar_meas_cov这个参数的含义是什么呢？什么时候该调大或者调小呢？

Joanna-HE, 2024-05-31:
你好，可以考虑当雷达点比较密集时，用较大的lidar_meas_cov。结构比较单一时，用较大的lidar_meas_cov。

以上是比较经验性的结论。
________________________________
寄件者: HaoRui ***@***.***>
寄件日期: 2024年5月29日 下午 04:28
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] use_imu_as_input参数的设置 (Issue #37)

好的！我已经尝试过修改lidar_meas_cov，发现效果有了很大的提升： [Screenshot from 2023-07-12 13-11-56] <https://private-user-images.githubusercontent.com/61641798/252861532-09f4c228-e09f-4768-9cf5-af434d2c6db2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5NzEzNzgsIm5iZiI6MTcxNjk3MTA3OCwicGF0aCI6Ii82MTY0MTc5OC8yNTI4NjE1MzItMDlmNGMyMjgtZTA5Zi00NzY4LTljZjUtYWY0MzRkMmM2ZGIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDA4MjQzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFhMjFkZDgwM2EyZmM4NTJiZjVjMmFkMzc2ZGEwN2UxNmFiN2M3OWQwOWJkMTI4NmEzMzcyNmFiMzAwODZkN2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZmIIcxPFbttEey5Dnuyrsq03j5PGbk_DZfQAK-EfAuk>  但是使用同一数据连续测试两次发现，在资源受限较高的情况下（比如第一次测试我开了其他程序导致CPU占用率比较高），lio的效果会有明显的下降（定位突变了一次），想问下程序里哪块会有随机性吗？ 还想请教一下，在特别开旷的场景下（比如机场，Lidar可能只能扫到地面），point-lio作为定位源来使用可靠程度高吗？ 我会尝试调整不同的参数来查看对比效果的好坏，非常感谢您的帮助～

你好，想问一下lidar_meas_cov这个参数的含义是什么呢？什么时候该调大或者调小呢？

hr2894235132, 2024-06-04:
> 你好，可以考虑当雷达点比较密集时，用较大的lidar_meas_cov。结构比较单一时，用较大的lidar_meas_cov。 以上是比较经验性的结论。
> […](#)

感谢回复！我去针对不同场景调整试一下。