hku-mars/Point-LIO #38: /

SSZ1, 2023-07-17:


SSZ1, 2023-07-19:
All right. Thanks a lot.

SSZ1, 2023-07-19:
> Please check the following two steps:
> 
> After launching the Point-LIO and playing the rosbag, use the command 'rostopic list' to show all the topics, and check if there are any topics published, and echo the topic with the right name. If there is no topic needed being published, check if the variable for publishing the according message is set as true in the '.yaml' file.

I have tried. But there is no variable about "/Laser_map" in yaml file.    And I saw the publisher of "/Laser_map" in cpp file was "publish_init_kdtree(pubLaserCloudMap);", but there is still no message when I echoed it.

Joanna-HE, 2023-07-19:
publish_init_kdtree only works for publishing the points of the initial kdtree, which means the /Laser_map will have message only at the beginning of the system run. After the kdtree is built, there would be no message in the topic /Laser_map. If you want to receive the message of point cloud during the running of the system, please echo /cloud_registered instead.
________________________________
寄件者: SUN SHANGZHE ***@***.***>
寄件日期: 2023年7月19日 下午 09:05
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38)

Please check the following two steps:

After launching the Point-LIO and playing the rosbag, use the command 'rostopic list' to show all the topics, and check if there are any topics published, and echo the topic with the right name. If there is no topic needed being published, check if the variable for publishing the according message is set as true in the '.yaml' file.

I have tried. But there is no variable about "/Laser_map" in yaml file. And I saw the publisher of "/Laser_map" in cpp file was "publish_init_kdtree(pubLaserCloudMap);", but there is still no message when I echoed it.

Joanna-HE, 2023-07-19:
Additionally, if you want to receive the message of /Laser_map, echo this topic before starting to play the rosbag.
________________________________
寄件者: ***@***.*** ***@***.***>
寄件日期: 2023年7月20日 上午 01:39
收件者: hku-mars/Point-LIO ***@***.***>
主旨: 回覆: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38)

publish_init_kdtree only works for publishing the points of the initial kdtree, which means the /Laser_map will have message only at the beginning of the system run. After the kdtree is built, there would be no message in the topic /Laser_map. If you want to receive the message of point cloud during the running of the system, please echo /cloud_registered instead.
________________________________
寄件者: SUN SHANGZHE ***@***.***>
寄件日期: 2023年7月19日 下午 09:05
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38)

Please check the following two steps:

After launching the Point-LIO and playing the rosbag, use the command 'rostopic list' to show all the topics, and check if there are any topics published, and echo the topic with the right name. If there is no topic needed being published, check if the variable for publishing the according message is set as true in the '.yaml' file.

I have tried. But there is no variable about "/Laser_map" in yaml file. And I saw the publisher of "/Laser_map" in cpp file was "publish_init_kdtree(pubLaserCloudMap);", but there is still no message when I echoed it.

SSZ1, 2023-07-20:
> Additionally, if you want to receive the message of /Laser_map, echo this topic before starting to play the rosbag.
> […](#)
> ________________________________ 寄件者: ***@***.*** ***@***.***> 寄件日期: [...]

Thanks  a lot. But I wanna get the /Laser_map like in fast_lio2, which is received by downsampling. So I wonder if I can get this by  add the code "publish_frame_world(pubLaserCloudMap)"  ?

Joanna-HE, 2023-07-20:
Yes, of course.
________________________________
寄件者: SUN SHANGZHE ***@***.***>
寄件日期: 2023年7月20日 上午 11:47
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38)

Additionally, if you want to receive the message of /Laser_map, echo this topic before starting to play the rosbag.
…
________________________________ 寄件者: @.*** @.> 寄件日期: 2023年7月20日 上午 01:39 收件者: hku-mars/Point-LIO @.> 主旨: 回覆: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38<https://github.com/hku-mars/Point-LIO/issues/38>) publish_init_kdtree only works for publishing the points of the initial kdtree, which means the /Laser_map will have message only at the beginning of the system run. After the kdtree is built, there would be no message in the topic /Laser_map. If you want to receive the message of point cloud during the running of the system, please echo /cloud_registered instead.
________________________________ 寄件者: SUN SHANGZHE @.> 寄件日期: 2023年7月19日 下午 09:05 收件者: hku-mars/Point-LIO @.> 副本: @.*** @.>; Comment @.> 主旨: Re: [hku-mars/Point-LIO] Topic "/Laser_map" has no message. (Issue #38<https://github.com/hku-mars/Point-LIO/issues/38>) Please check the following two steps: After launching the Point-LIO and playing the rosbag, use the command 'rostopic list' to show all the topics, and check if there are any topics published, and echo the topic with the right name. If there is no topic needed being published, check if the variable for publishing the according message is set as true in the '.yaml' file. I have tried. But there is no variable about "/Laser_map" in yaml file. And I saw the publisher of "/Laser_map" in cpp file was "publish_init_kdtree(pubLaserCloudMap);", but there is still no message when I echoed it.

Thanks a lot. But I wanna get the /Laser_map like in fast_lio2, which is received by downsampling. So I wonder if I can get this by add the code "publish_frame_world(pubLaserCloudMap)" ?

SSZ1, 2023-07-20:
> Yes, of course.
> […](#)
> ________________________________ 寄件者: SUN SHANGZHE ***@***.***> 寄件日期: 2023年7月20日 上午 11:47 收件者: hku-mars/Point-LIO ***@***.***> 副本: ***@***.*** ***@***.***>; Comment [...]

Oh~Thanks for your quick reply.