hku-mars/Point-LIO #40: 点阵式地图，地图构建异常

yuangusshuai, 2023-08-04:
请问我的地图构建为什么会是这种？看起来像点阵式一样，我使用的mid360，这是什么问题呢？
[image]

Joanna-HE, 2023-08-04:
你好，点云的downsample参数，你采用的是多少呢？看起来像是空间降采样的resolution太小了，所以每个空间格子只有一个点。
________________________________
寄件者: yuangusshuai ***@***.***>
寄件日期: 2023年8月4日 上午 11:08
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 点阵式地图，地图构建异常 (Issue #40)

请问我的地图构建为什么会是这种？看起来像点阵式一样，我使用的mid360，这是什么问题呢？
[2023-08-04 11-03-12 的屏幕截图]<https://user-images.githubusercontent.com/90245832/258285079-e78e898a-b61e-4ac9-a2f7-c33ec76726f7.png>

yuangusshuai, 2023-08-04:
> 你好，点云的downsample参数，你采用的是多少呢？看起来像是空间降采样的resolution太小了，所以每个空间格子只有一个点。
> […](#)
感谢您的及时回复，没有找到点云的downsample相关参数，请问应该修改哪里呢，希望得到指导，另外在跑数据集时没有出现这种情况，这是否与雷达ros包有关？非常感谢您的再次回复！

Joanna-HE, 2023-08-04:
downsample的参数在launch文件中，filter_size_map对应的数值越大，downsample越狠。这种点阵情况和rosbag是有关系的，而且它的出现是正常的。
________________________________
寄件者: yuangusshuai ***@***.***>
寄件日期: 2023年8月4日 下午 01:25
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 点阵式地图，地图构建异常 (Issue #40)

你好，点云的downsample参数，你采用的是多少呢？看起来像是空间降采样的resolution太小了，所以每个空间格子只有一个点。
…
________________________________ 寄件者: yuangusshuai @.> 寄件日期: 2023年8月4日 上午 11:08 收件者: hku-mars/Point-LIO @.> 副本: Subscribed @.> 主旨: [hku-mars/Point-LIO] 点阵式地图，地图构建异常 (Issue #40<https://github.com/hku-mars/Point-LIO/issues/40>) 请问我的地图构建为什么会是这种？看起来像点阵式一样，我使用的mid360，这是什么问题呢？ [2023-08-04 11-03-12 的屏幕截图]https://user-images.githubusercontent.com/90245832/258285079-e78e898a-b61e-4ac9-a2f7-c33ec76726f7.png D
感谢您的及时回复，没有找到点云的downsample相关参数，请问应该修改哪里呢，希望得到指导，另外在跑数据集时没有出现这种情况，这是否与雷达ros包有关？非常感谢您的再次回复！