hku-mars/Point-LIO #42: 建图累积偏移

yangtzey, 2023-08-11:
首先非常感谢您的开源,我在适配补盲激光雷达的时候发现一个问题,使用output模式，感觉收敛的很慢，导致累积偏移很大，导致建图漂移，如下图表现
[image]
我的配置文件如下，需要再从哪些方向调整参数。非常感谢
common:
    lid_topic: "/raw_points" 
    imu_topic: "/imu" 
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency   
    time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                                # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 5
    scan_line: 32
    timestamp_unit: 2           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 2.0 

mapping:
    imu_en: true
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.01 # = 1 / frequency of IMU
    satu_acc: 30.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.80 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.1 # 0.001
    acc_cov_output: 500
    gyr_cov_output: 500 
    b_acc_cov: 0.01 
    b_gyr_cov: 0.001
    imu_meas_acc_cov: 0.2 #0.1 # 2
    imu_meas_omg_cov: 0.2 #0.1 # 2 
    gyr_cov_input: 0.001 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 120 
    det_range: 100.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [ 0, 0.0, 0.06] 
    extrinsic_R: [ 0, -1, 0,
                   1, 0, 0,
                   0, 0, 1 ] # ulhk 4 utbm 
odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

Joanna-HE, 2023-08-11:
你好，感谢你对我们工作的关注。尝试把gyr_cov_output: 改回1000，并且减小imu_meas_acc_cov为0.1，imu_meas_omg_cov为0.01，试下呢？
________________________________
寄件者: yangtzey ***@***.***>
寄件日期: 2023年8月11日 下午 01:32
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 建图累积偏移 (Issue #42)

首先非常感谢您的开源,我在适配补盲激光雷达的时候发现一个问题,使用output模式，感觉收敛的很慢，导致累积偏移很大，导致建图漂移，如下图表现
[Selection_029]<https://user-images.githubusercontent.com/6038552/259928319-866a4bea-c11b-4c4b-98c0-bc038e9cabe4.png>
我的配置文件如下，需要再从哪些方向调整参数。非常感谢
common:
lid_topic: "/raw_points"
imu_topic: "/imu"
con_frame: false # true: if you need to combine several LiDAR frames into one
con_frame_num: 1 # the number of frames combined
cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency
time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
# the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
lidar_type: 5
scan_line: 32
timestamp_unit: 2 # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
blind: 2.0

mapping:
imu_en: true
extrinsic_est_en: false # for aggressive motion, set this variable false
imu_time_inte: 0.01 # = 1 / frequency of IMU
satu_acc: 30.0 # the saturation value of IMU's acceleration. not related to the units
satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
acc_norm: 9.80 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
lidar_meas_cov: 0.1 # 0.001
acc_cov_output: 500
gyr_cov_output: 500
b_acc_cov: 0.01
b_gyr_cov: 0.001
imu_meas_acc_cov: 0.2 #0.1 # 2
imu_meas_omg_cov: 0.2 #0.1 # 2
gyr_cov_input: 0.001 # for IMU as input model
acc_cov_input: 0.1 # for IMU as input model
plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
match_s: 81
fov_degree: 120
det_range: 100.0
gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
gravity: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # gravity to be aligned
gravity_init: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
extrinsic_T: [ 0, 0.0, 0.06]
extrinsic_R: [ 0, -1, 0,
1, 0, 0,
0, 0, 1 ] # ulhk 4 utbm
odometry:
publish_odometry_without_downsample: false

publish:
path_en: true # false: close the path output
scan_publish_en: true # false: close all the point cloud output
scan_bodyframe_pub_en: false # true: output the point cloud scans in IMU-body-frame

pcd_save:
pcd_save_en: false
interval: -1 # how many LiDAR frames saved in each pcd file;
# -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

yangtzey, 2023-08-11:
这是来自QQ邮箱的假期自动回复邮件。
自动回复

yangtzey, 2023-08-15:
非常感谢您这么快的回答，按照您的建议把配置文件修改如下：
common:
    lid_topic: "/raw_points" 
    imu_topic: "/imu" 
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency   
    time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                                # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 5
    scan_line: 32
    timestamp_unit: 2           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.

mapping:
    imu_en: true
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.01 # = 1 / frequency of IMU
    satu_acc: 30.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.80 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.1 # 0.001
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.01 
    b_gyr_cov: 0.001
    imu_meas_acc_cov: 0.1 #0.1 # 2
    imu_meas_omg_cov: 0.01 #0.1 # 2 
    gyr_cov_input: 0.001 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.05 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 120 
    det_range: 100.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.80] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [ 0, 0.0, 0.06] # ulhk # [-0.5, 1.4, 1.5] # utbm
    # extrinsic_R: [ 0, 1, 0,
    #                -1, 0, 0,
    #                0, 0, 1 ] # ulhk 5 6
    # extrinsic_R: [ 0, -1, 0,
    #                 1, 0, 0,
    #                 0, 0, 1 ] # utbm 1, 2
    extrinsic_R: [ 0, -1, 0,
                   1, 0, 0,
                   0, 0, 1 ] # ulhk 4 utbm 
odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
效果如下，有较大改善但是还是没有闭合上，还可以从调整那些参数。非常感谢
[image]
另外一个问题，我使用的补盲雷达 只能扫描前向25m的范围，有的时候会出现大面积扫描地面和一面平墙情况，特征出现退化。这个时候建图的点云会变厚，并修改了共面阈值plane_thr: 0.05 ，代码中修改了查找最近点由5个变更成6个，效果如下：
[image]
针对这个问题还有什么比较好的解决方案，非常感谢您的回答

closed 2023-10-18 by yangtzey