hku-mars/Point-LIO #44: 屏蔽雷达周围点

IMUTsongX, 2023-08-17:
你好，我们测试您的算法遇到这样一个问题，我们通过修改blind的值，以及初始化blind(0.01)的值，都没有起到雷达点云的屏蔽作用，按理说修改blind的值，点云会自动屏蔽以blind为半径的点，但是无论我们怎样修改，都没见到任何效果，可以告知原因吗；但是对于fast lio，修改blind，能起到很好的屏蔽作用

Joanna-HE, 2023-08-17:
请问你用的livox的雷达吗？我刚刚查到livox的preprocessor漏写了一个括号，导致blind不起作用。其他的preprocessor应该没什么问题。我将修改后的livox的preprocessor上传在新的commit里了，可以尝试下是否还有问题。
________________________________
寄件者: ZhangSS ***@***.***>
寄件日期: 2023年8月17日 下午 03:31
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 屏蔽雷达周围点 (Issue #44)

你好，我们测试您的算法遇到这样一个问题，我们通过修改blind的值，以及初始化blind(0.01)的值，都没有起到雷达点云的屏蔽作用，按理说修改blind的值，点云会自动屏蔽以blind为半径的点，但是无论我们怎样修改，都没见到任何效果，可以告知原因吗；但是对于fast lio，修改blind，能起到很好的屏蔽作用

IMUTsongX, 2023-08-17:
很感谢您能回复，我们使用的livox MID360 修改后的livox的preprocessor上传在新的commit里，但是我们并没有发现你更新的新代码，是否有其他的代码存放地址呢，非常感谢

IMUTsongX, 2023-08-17:
你好，我们已找到问题，已经修改成功，感谢您的回复

G4419, 2023-10-23:
> 你好，我们已找到问题，已经修改成功，感谢您的回复

你好，请问你是怎么解决的，我也遇到了这个问题

G4419, 2023-10-27:
不好意思，代码没问题。