hku-mars/Point-LIO #49: Is there any tips for setting saturation value?

Taeyoung96, 2023-08-30:
@Joanna-HE

Thanks for sharing a great project! :smile:  
I have some questions about setting parameters in the config file.

I understand that saturation values are used in the `h_model_IMU_output()` function.
Is there any tip to set these values when applying to a custom dataset?
What kind of information should I look at from the IMU to determine the saturation value?

Also, I understand that acc_cov_output and gyr_cov_output are related to the noise covariance matrix.
What is the impact on Odometry if these values are large?
Is there a reason why 500 and 1000 are the defaults?

Thanks again for sharing awesome work, and I look forward to your response.

Joanna-HE, 2023-09-04:
Thanks very much for your interest in our work!

The saturation value of an IMU is equal to the absolute value of its measuring range. And you should also pay attention to the measuring units of the IMU, since some IMUs (such as the built-in IMU of Livox LiDARs) use 'g' (=9.8 m/s^2) as the units of accelerators.

When the acc_cov_output and gyr_cov_output are large, the Point-LIO is also able to work with slightly different performances. It is your option to set the values of those variables. 500 and 1000 are choosen since they are enough to stimulate the variance of acceleration and angular velocity in practice.

Best regards,

Dongjiao He
________________________________
寄件者: TaeYoung Kim ***@***.***>
寄件日期: 2023年8月30日 下午 11:07
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Mention ***@***.***>
主旨: [hku-mars/Point-LIO] Is there any tips for setting saturation value? (Issue #49)

@Joanna-HE<https://github.com/Joanna-HE>

Thanks for sharing a great project! 😄
I have some questions about setting parameters in the config file.

I understand that saturation values are used in the h_model_IMU_output() function.
Is there any tip to set these values when applying to a custom dataset?
What kind of information should I look at from the IMU to determine the saturation value?

Also, I understand that acc_cov_output and gyr_cov_output are related to the noise covariance matrix.
What is the impact on Odometry if these values are large?
Is there a reason why 500 and 1000 are the defaults?

Thanks again for sharing awesome work, and I look forward to your response.

Taeyoung96, 2023-09-04:
@Joanna-HE   
Thank you for your kind response!  

I have a question about the absolute value of its measuring range.
If I plot the IMU angular velocity I acquired, it looks like below.
[image]

If I don't know the IMU Specifications, can I get the measuring range and `satu_gyro` with the above measurements?

Joanna-HE, 2023-10-22:
It looks like that the measuring limit has not been reached by the motions. And as indicated in the figure, the saturation (i.e., the absolute value of  measuring  range) is larger than 2.5 rad/s.

Taeyoung96, 2023-10-25:
@Joanna-HE   
Thanks for your reply! 
How will the data be visualized when the measuring limit is reached?
Do you mean when the measurement fails above some constant value?

closed 2024-05-24 by Taeyoung96