hku-mars/Point-LIO #51: wrong Odometry

giangdao1402, 2023-09-30:
I use imu of pixhawk 6C, Ouster lidar 16 mounted upside down. so i change the extrinsic_R of Z to visualize with +Z in rviz. But my Odometry data seem odd. Can you please take a look at it for me.                                                                                                                                                                                                                                                    common:
    lid_topic:  "/ouster/points"
    #imu_topic:  "/ouster/imu"
    #imu_topic: "/imu_data"
    imu_topic: "/mavros/imu/data_raw"
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency 
    time_lag_imu_to_lidar: 1684841499.635005 #1684841493.84092665 #1684841499.635005 #1684841524.65910959  #1684841499.635005 #1694165984.83338928 #1694168642.17952871 #0.006376 #-0.48057733 #1684841524.65910959 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 3 # 1 Livox Avia LiDAR
    scan_line: 16 # 32 #velodyne 6 avia
    timestamp_unit: 3           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 4

mapping:
    imu_en: true
    start_in_aggressive_motion: false # if true, a preknown gravity should be provided in following gravity_init
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.02 # = 1 / frequency of IMU
    satu_acc: 30 #80.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 25 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.81 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.01 # 0.01
    acc_cov_output: 1000
    gyr_cov_output: 2000 
    b_acc_cov: 0.07 
    b_gyr_cov: 0.3 
    imu_meas_acc_cov: 0.1 #0.1 # 2
    imu_meas_omg_cov: 0.1 #0.1 # 2 
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 300.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity:  [0.0, 0.0, -9.810]  #[3.580628, 0.025035, -9.133134] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [0.0, 0.0, 0.0] #[-0.160931, 0.337930, -0.054396] #[-0.277508, -0.179171, -0.014018] #[0.0, 0.0, 0.0]ssss
    extrinsic_R:  [1, 0, 0,
                   0, -1, 0,
                   0, 0, -1 ] 
                  #[ 0.033039, -0.822540, 0.567746,
                   #-0.987192, -0.115566, -0.109983,
                    #0.156078 , -0.556841, -0.815824 ]

odometry: 
[image]

    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file;

FPSychotic, 2023-09-30:
Probably your mavros odometry is in NED, ROS use ENU, just  is the first thing I would check, later the lidar-imu matrix is right. Some posts here in issues about it, just check that issues to be familiar with some things

Joanna-HE, 2023-10-22:
Hi, as indicated in the comments of FPSychotic, the problem is caused by the wrong alignment of the ground truth and the output odometry of Point-LIO, the odometry of Point-LIO takes the start point as the origin point, which may be not the same as that of the origin point of the ground truth, please align the trajectories before comparing the odometry of Point-LIO with ground truth.