hku-mars/Point-LIO #55: Airsim problem

giangdao1402, 2023-10-08:
I use Airsim to simulate Lidar (Ouster) point cloud. Because Airsim's lidar only give x,y,z of point cloud so  i write code to have x,y,z and intensity. When i use LiDAR_IMU_Init package rviz show this 
[image]
But when i use point_lio rviz show this 
[image]
I can't see what is missing, it seems odd. Can anyone provide some ideas on how to solve this problem?

johnnychen412, 2023-10-11:
What kind of problems are you having? Can you show me the .yaml file of point-lio and the setting.json of airsim?

giangdao1402, 2023-10-15:
point_lio only show red color (i dont know what is means). And there is no path (topic /aft_mapped_to_init show nothing). `common:
    #lid_topic:  "/velodyne_points"
    #imu_topic:  "/ouster/imu"
    #lid_topic:   "/ouster/points"
    lid_topic:   "/airsim_node/drone_1/lidar/LidarCustom"
    #imu_topic:   "/camera/imu"
    imu_topic: "/airsim_node/drone_1/imu/Imu"
    #imu_topic: "/mavros/imu/data_raw"
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency 
    time_lag_imu_to_lidar: -9.36584783 #0.078602 #-1225.06527281 #1684841499.635005 #1684841493.84092665 #1684841499.635005 #1684841524.65910959  #1684841499.635005 #1694165984.83338928 #1694168642.17952871 #0.006376 #-0.48057733 #1684841524.65910959 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 3 # 1 Livox Avia LiDAR
    scan_line: 16 # 32 #velodyne 6 avia
    timestamp_unit: 3           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1

mapping:
    imu_en: true
    start_in_aggressive_motion: false # if true, a preknown gravity should be provided in following gravity_init
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.005 # = 1 / frequency of IMU
    satu_acc: 3 #80.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 25 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.81 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.01 # 0.01
    acc_cov_output: 1000
    gyr_cov_output: 2000 
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    imu_meas_acc_cov: 0.012 #0.1 # 2
    imu_meas_omg_cov: 0.05  #0.1 # 2 
    gyr_cov_input: 0.005 # for IMU as input model
    acc_cov_input:  0.04 # for IMU as input model
    plane_thr: 0.05 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 50.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity:  [0.0, 0.0, -9.810]  #[3.580628, 0.025035, -9.133134] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [0, 0, 0] #[-0.160931, 0.337930, -0.054396] #[-0.277508, -0.179171, -0.014018] #[0.0, 0.0, 0.0]ssss
    extrinsic_R:  [1, 0, 0,
                   0, 1, 0,
                   0, 0, 1]
odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.``{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true,
          "AngularRandomWalk": 0.3,
          "GyroBiasStabilityTau": 500,
          "GyroBiasStability": 4.6,
          "VelocityRandomWalk": 0.24,
          "AccelBiasStabilityTau": 800,
          "AccelBiasStability": 36
        },        
        "LidarCustom": {
          "SensorType": 6,
          "Range": 100,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 327680,
          "RotationsPerSecond": 10,
          "VerticalFOVUpper": 16.5,
          "VerticalFOVLower": -16.5,
          "HorizontalFOVStart": -180,
          "HorizontalFOVEnd": 180,
          "X": 2, "Y": 0, "Z": -1,
          "DrawDebugPoints": true,
          "Pitch":0, "Roll": 0, "Yaw": 0,
          "DataFrame": "SensorLocalFrame"
        }
      }
    }
  }
}`

Joanna-HE, 2023-10-22:
There is no problem actually. The different color of LIDAR_IMU_INIT and Point-LIO is caused by the different parameter settings for the appearance of the map points, you can change those parameters to any values you like in the publish function of the code.

giangdao1402, 2023-10-23:
Well,topic odometry /aft_mapped_to_init not publish, and the map only shows when bag run over half length. Can you give me an advise ?

Joanna-HE, 2023-10-23:
Have no direct idea why the topic odometry/aft_mapped_to_init not publish, check if the function "publish_odometry()" is called or not.

For "The map only shows when bag run over half length", dose the time between LiDAR and IMU be sync?
________________________________
寄件者: Giang Dao ***@***.***>
寄件日期: 2023年10月23日 下午 02:52
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] Airsim problem (Issue #55)

Well,topic odometry /aft_mapped_to_init not publish, and the map only shows when bag run over half length. Can you give me an advise ?

giangdao1402, 2023-10-23:
yes of course, the imu initialize 100% right after i play the bag file 
[image]