hku-mars/Point-LIO #59: 一个mid360的rosbag表现不佳

jiejie567, 2023-11-04:
你好，Dr. He。我正在做一个基于point lio的应用的工作。我自己录制了几个mid360的数据集，静止3s以上，然后开始转动，在用imu模式或者不用imu （及input 和 output模式）点云都比较模糊或者崩溃。让身边的几个师弟运行也都是这样的结果，请问大概是什么原因？（我用fast lio运行,结果良好）假如方便，我可以发你一份我的数据。
再次感谢优秀的工作。

Joanna-HE, 2023-11-04:
你好，感谢您对这项工作的兴趣。可以先尝试如下的操作：

  1.  确认使用的是ubuntu20.04，而不是18.04，point-lio在18.04上会触发一些导致崩溃的问题。
  2.  确认IMU的参数和外参准确地写在yaml文件中了。

如果上面都没办法解决问题的话，可以发我你们一个数据，我找下问题在哪里。

祝好，

贺东娇
________________________________
寄件者: jiejie567 ***@***.***>
寄件日期: 2023年11月4日 下午 06:27
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 一个mid360的rosbag表现不佳 (Issue #59)

你好，Dr. He。我正在做一个基于point lio的应用的工作。我自己录制了几个mid360的数据集，静止3s以上，然后开始转动，在用imu模式或者不用imu （及input 和 output模式）点云都比较模糊或者崩溃。让身边的几个师弟运行也都是这样的结果，请问大概是什么原因？（我用fast lio运行,结果良好）假如方便，我可以发你一份我的数据。
再次感谢优秀的工作。

jiejie567, 2023-11-06:
谢谢贺博。
你说的几个可能会产生问题的原因我们排除了，我们也阅读了其他的issue。当前已发送rosbag。
再次感谢贺博的优秀工作和热心地提供支持。Point-LIO这个工作给了我们很大的帮助。：）
徐杰

Joanna-HE, 2023-11-10:
你好，

我刚刚看了下，然后知道问题在哪里了，SO(3) 用quaternion表示，在和rotation之间转换的时候会出现行/列向量的模长不为1的情况。可以把代码中的SO（3）全部替换成用rotation表示的形式。我明后两天会看看能不能改完然后传一份新的代码上来，你们也可以试试自行改一下~

祝好，

贺东娇
________________________________
寄件者: jiejie567 ***@***.***>
寄件日期: 2023年11月6日 上午 10:19
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 一个mid360的rosbag表现不佳 (Issue #59)

谢谢贺博。
你说的几个可能会产生问题的原因我们排除了，我们也阅读了其他的issue。当前已发送rosbag。
再次感谢贺博的优秀工作和热心地提供支持。Point-LIO这个工作给了我们很大的帮助。：）
徐杰

Joanna-HE, 2023-11-11:
你好，你可以试下新的commit，然后调整一下地图参数和imu_meas_omg_cov（比如1.0），效果会好一些。但没有从根本上解决。想把这个bag跑得好，需要更改地图，把目前的ikdtree替换成faster-lio的voxel map，这个问题会被彻底解决。
________________________________
寄件者: ***@***.*** ***@***.***>
寄件日期: 2023年11月11日 上午 03:07
收件者: hku-mars/Point-LIO ***@***.***>
主旨: 回覆: [hku-mars/Point-LIO] 一个mid360的rosbag表现不佳 (Issue #59)

你好，

我刚刚看了下，然后知道问题在哪里了，SO(3) 用quaternion表示，在和rotation之间转换的时候会出现行/列向量的模长不为1的情况。可以把代码中的SO（3）全部替换成用rotation表示的形式。我明后两天会看看能不能改完然后传一份新的代码上来，你们也可以试试自行改一下~

祝好，

贺东娇
________________________________
寄件者: jiejie567 ***@***.***>
寄件日期: 2023年11月6日 上午 10:19
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 一个mid360的rosbag表现不佳 (Issue #59)

谢谢贺博。
你说的几个可能会产生问题的原因我们排除了，我们也阅读了其他的issue。当前已发送rosbag。
再次感谢贺博的优秀工作和热心地提供支持。Point-LIO这个工作给了我们很大的帮助。：）
徐杰

jiejie567, 2023-11-12:
非常感谢！

Joanna-HE, 2024-01-18:
你好，我最近完成了地图的更改，并发布在了”point-lio-with-grid-map“这个branch上，可以解决你的bag的问题~

johnnychen412, 2024-07-02:
你好，新版本能运行NTU VIRAL的数据集吗？