hku-mars/Point-LIO #70: 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大

Joweay, 2024-02-21:
采用livox-mid360雷达录制室内多层数据集，Point-LIO定位结果漂移，但fast-lio2可以较好的定位。
参数配置如下：
[image]
[数据集如下](https://1drv.ms/u/c/7288ccf48d538012/EQUcLgHaCrNMj5Mx9FCCzywBDDUjbFAzMYMUQ0SbIf0Myw?e=IqeA0i)

Joweay, 2024-02-21:
@Joanna-HE 感谢及时回复，用的是最新的poinglio-grid_map代码，调整了很多参数，都不能顺利跑出来

Joanna-HE, 2024-02-21:
把satu_acc改成30.0, acc_norm改成9.81。其他参数采用分支中的默认参数即可。上述两个数值与数据包采集时使用的IMU设定的量程有关。
________________________________
寄件者: Joweay ***@***.***>
寄件日期: 2024年2月21日 下午 05:47
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Mention ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大 (Issue #70)

@Joanna-HE<https://github.com/Joanna-HE> 感谢及时回复，用的是最新的poinglio-grid_map代码，调整了很多参数，都不能顺利跑出来

Joanna-HE, 2024-02-22:
然后再把imu_meas_acc_cov改成0.01，plane_tr 改为0.05，效果会更好一些。
________________________________
寄件者: ***@***.*** ***@***.***>
寄件日期: 2024年2月22日 上午 02:01
收件者: hku-mars/Point-LIO ***@***.***>
主旨: 回覆: [hku-mars/Point-LIO] 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大 (Issue #70)

把satu_acc改成30.0, acc_norm改成9.81。其他参数采用分支中的默认参数即可。上述两个数值与数据包采集时使用的IMU设定的量程有关。
________________________________
寄件者: Joweay ***@***.***>
寄件日期: 2024年2月21日 下午 05:47
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Mention ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大 (Issue #70)

@Joanna-HE<https://github.com/Joanna-HE> 感谢及时回复，用的是最新的poinglio-grid_map代码，调整了很多参数，都不能顺利跑出来

Joweay, 2024-02-22:
> 把satu_acc改成30.0, acc_norm改成9.81。其他参数采用分支中的默认参数即可。上述两个数值与数据包采集时使用的IMU设定的量程有关。
> […](#)
感谢，效果有所改进，但是在下楼时数据飘的还是很大，如下图
[image]

Joanna-HE, 2024-02-22:
satu_acc:30.0, acc_norm:9.8,plane_thr:0.05, imu_meas_acc_cov:0.01。我这边测试是没有发生漂移的。再检查一下你的配置细节呢？
________________________________
寄件者: Joweay ***@***.***>
寄件日期: 2024年2月22日 上午 11:12
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Mention ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大 (Issue #70)

把satu_acc改成30.0, acc_norm改成9.81。其他参数采用分支中的默认参数即可。上述两个数值与数据包采集时使用的IMU设定的量程有关。
…
________________________________ 寄件者: Joweay @.> 寄件日期: 2024年2月21日 下午 05:47 收件者: hku-mars/Point-LIO @.> 副本: @.*** @.>; Mention @.> 主旨: Re: [hku-mars/Point-LIO] 基于livox-mid360雷达，Point-LIO在室内多层楼梯数据集上位姿估计漂移极大 (Issue #70<https://github.com/hku-mars/Point-LIO/issues/70>) @Joanna-HE<https://github.com/Joanna-HE>https://github.com/Joanna-HE 感谢及时回复，用的是最新的poinglio-grid_map代码，调整了很多参数，都不能顺利跑出来 D
感谢，效果有所改进，但是在下楼时数据飘的还是很大，如下图
image.png (view on web)<https://github.com/hku-mars/Point-LIO/assets/24538792/bb204fed-9e40-4235-af85-5f4d00fe36ba>

Joweay, 2024-02-22:
> satu_acc:30.0, acc_norm:9.8,plane_thr:0.05, imu_meas_acc_cov:0.01。我这边测试是没有发生漂移的。再检查一下你的配置细节呢？
> […](#)
> ________________________________ 寄件者: Joweay ***@***.***> 寄件日期: 2024年2月22日 上午 11:12 收件者: [...]

感谢贺博，我的问题在于使用了"use_imu_as_input"导致把局部坐标系下点云变换到全局坐标系下误差过大，关闭该参数后，但是整个过程中end-to-end累计误差较大，我跑出来的估计如下图，蓝色线条为大概的误差值
[image]