hku-mars/Point-LIO #74: [laserMapping-1] process has died (exit code -11)

alexandre-bonnefond, 2024-03-13:
Hello, 
I am using your code with the Livox MID-360 and livox_ros_driver2 ([git](https://github.com/Livox-SDK/livox_ros_driver2?tab=readme-ov-file)). It requires some modifications of your code that I found here [see livox_ros_driver2 branch](https://github.com/ctu-mrs/Point-LIO.git). The lidar works well and I am able to read the topics from the ros_driver2 but I have some issues with Point-Lio. Most of the time i am getting an error (see picture) and nothing appears on rviz. The fact is that sometimes it works without any modification of the code and I get the desired output. I know this is a long shot as it is not using your orginal code but would you have any idea regarding why I get this error ?

Thank you 

[image]

Hezhexi2002, 2024-04-11:
> Hello, I am using your code with the Livox MID-360 and livox_ros_driver2 ([git](https://github.com/Livox-SDK/livox_ros_driver2?tab=readme-ov-file)). It requires some modifications of your code that I found here [see livox_ros_driver2 branch](https://github.com/ctu-mrs/Point-LIO.git). The lidar works well and I am able to read the topics from the ros_driver2 but I have some issues with Point-Lio. Most of the time i am getting an error (see picture) and nothing appears on rviz. The fact is that sometimes it works without any modification of the code and I get the desired output. I know this is a long shot as it is not using your orginal code but would you have any idea regarding why I get this error ?
> 
> Thank you
> 
> ![Screenshot from 2024-03-13 10-55-36](https://private-user-images.githubusercontent.com/45358515/312406871-b63bf048-dc2e-4688-8f29-3fb6e8842c1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4NDAzNDQsIm5iZiI6MTcxMjg0MDA0NCwicGF0aCI6Ii80NTM1ODUxNS8zMTI0MDY4NzEtYjYzYmYwNDgtZGMyZS00Njg4LThmMjktM2ZiNmU4ODQyYzFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDEyNTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzMTQ0NTlmOWE1Y2ExZTA3ODQyYTM1ODZjNmZlMzk0ZTljYjkzOWJjNGYzZmNmNzYzM2M5ZmQzMWE4YmMxOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vAFheTw1KZqmmstlb7waSn_9wIfVx1Ta6Hd7as0WjqI)

> Hello, I am using your code with the Livox MID-360 and livox_ros_driver2 ([git](https://github.com/Livox-SDK/livox_ros_driver2?tab=readme-ov-file)). It requires some modifications of your code that I found here [see livox_ros_driver2 branch](https://github.com/ctu-mrs/Point-LIO.git). The lidar works well and I am able to read the topics from the ros_driver2 but I have some issues with Point-Lio. Most of the time i am getting an error (see picture) and nothing appears on rviz. The fact is that sometimes it works without any modification of the code and I get the desired output. I know this is a long shot as it is not using your orginal code but would you have any idea regarding why I get this error ?
> 
> Thank you
> 
> ![Screenshot from 2024-03-13 10-55-36](https://private-user-images.githubusercontent.com/45358515/312406871-b63bf048-dc2e-4688-8f29-3fb6e8842c1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4NDAzNDQsIm5iZiI6MTcxMjg0MDA0NCwicGF0aCI6Ii80NTM1ODUxNS8zMTI0MDY4NzEtYjYzYmYwNDgtZGMyZS00Njg4LThmMjktM2ZiNmU4ODQyYzFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDEyNTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzMTQ0NTlmOWE1Y2ExZTA3ODQyYTM1ODZjNmZlMzk0ZTljYjkzOWJjNGYzZmNmNzYzM2M5ZmQzMWE4YmMxOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vAFheTw1KZqmmstlb7waSn_9wIfVx1Ta6Hd7as0WjqI)

@alexandre-bonnefond，Hello
I am currently facing the same issue that you described in your GitHub issue. I am also using the Livox MID-360 and livox_ros_driver2 (git) with the necessary modifications to the code found in the livox_ros_driver2 branch. Like you, I can successfully read the topics from the ros_driver2, but I encounter problems with Point-Lio.

Most of the time, I receive an error 
[image]
, and nothing appears in rviz. However, there are rare instances where the code works without any modifications, and I get the desired output.

I wanted to reach out and ask if you have managed to find a solution to this issue since posting about it. If you have any insights or suggestions on how to resolve this error, I would greatly appreciate your help.

Thank you for sharing your experience and for any assistance you can provide.

alexandre-bonnefond, 2024-04-12:
Hello @Hezhexi2002, 

Yes I actually managed to make it work. After investigating the code I figured that the process died when entering a specific condition somehow related to the IMU. I believe this explained why sometimes it worked, if the IMU was in a very specific configuration. 
Anyway, I also changed the config file for the MID360 and these solved everything. I didn't have time to clearly identify which parameter makes it work but I will do it soon. Here is the config file. I hope this will work for you too, let me know if not.

```
common:
    lid_topic:  "/livox/lidar" 
    imu_topic:  "/livox/imu" 
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency
    time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                           # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 1 
    scan_line: 6
    timestamp_unit: 1           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1.0 

mapping:
    imu_en: true
    extrinsic_est_en: true # for aggressive motion, set this variable false
    imu_time_inte: 0.005 # = 1 / frequency of IMU
    satu_acc: 3.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 1.0 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.001 # 0.001; 0.01
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.0001 
    b_gyr_cov: 0.0001 
    imu_meas_acc_cov: 0.1 #0.1 # 0.1
    imu_meas_omg_cov: 0.1 #0.01 # 0.1
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 50.0
    gravity_align: true # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    extrinsic_T: [ 0.011, 0.02329, -0.04412 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ]

odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

Hezhexi2002, 2024-04-18:
Dear @alexandre-bonnefond 

I apologize for not getting back to you sooner regarding the issue with Point LIO. After encountering the problem, I decided to switch to Fast LIO as a temporary solution.

Thank you for sharing the updated config file. I just tried it out, but unfortunately, I am still facing the same issue as before. Please find the attached screenshot from rviz illustrating the problem.

[image]
[image]

For your reference, here is my current launch file:

```
<launch>
<!-- Launch file for Livox Horizon LiDAR -->

	<arg name="rviz" default="true" />

	<node pkg="point_lio" type="pointlio_mapping" name="laserMapping" output="screen">
	<rosparam command="load" file="$(find point_lio)/config/mid360.yaml" />
	<param name="use_imu_as_input" type="bool" value="0"/> <!--change to 1 to use IMU as input of Point-LIO-->
	<param name="prop_at_freq_of_imu" type="bool" value="1"/>
	<param name="check_satu" type="bool" value="1"/>
	<param name="init_map_size" type="int" value="10"/>
	<param name="point_filter_num" type="int" value="3"/> <!--1, 3--> 
	<param name="space_down_sample" type="bool" value="1" />  
	<param name="filter_size_surf" type="double" value="0.5" /> <!--0.5, 0.3, 0.2, 0.15, 0.1--> 
	<param name="filter_size_map" type="double" value="0.5" /> <!--0.5, 0.3, 0.15, 0.1-->
	<param name="cube_side_length" type="double" value="1000" /> <!--1000-->
	<param name="runtime_pos_log_enable" type="bool" value="0" /> <!--1-->
	</node>
	<group if="$(arg rviz)">
	<node launch-prefix="nice" pkg="rviz" type="rviz" name="rviz" args="-d $(find point_lio)/rviz_cfg/loam_livox.rviz" />
	</group>

	launch-prefix="gdb -ex run --args"

</launch>
```

If you have any further insights or suggestions on how to resolve this issue, I would greatly appreciate your help. Thank you for your time and assistance.

Best regards,
hezhexi2002

JiaoxianDu, 2024-06-17:
Thanks @alexandre-bonnefond 
The config file works fine to me.