hku-mars/Point-LIO #76: 里程计输出延时问题

sarallonly, 2024-03-21:
请问里程计输出延时大约半分钟，这种问题应该怎么解决？
期待您的回答

dongjq23, 2024-03-22:
我也遇到了同样的问题，/aft_mapped_to_init话题相对播放的包有半分钟延迟。需要怎么解决一下呀

sarallonly, 2024-03-22:
现在还没有解决呢，之前怀疑是电脑的问题，换了个电脑发现这个问题还是存在

Joanna-HE, 2024-03-22:
其他话题有延迟吗？
________________________________
寄件者: sarallonly ***@***.***>
寄件日期: 2024年3月22日 下午 02:02
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 里程计输出延时问题 (Issue #76)

现在还没有解决呢，之前怀疑是电脑的问题，换了个电脑发现这个问题还是存在

sarallonly, 2024-03-22:
输出的点云也有延迟

 其他话题有延迟吗？ 
 寄件者: sarallonly ***@***.***&gt; 

 收件者: hku-mars/Point-LIO ***@***.***&gt; 
 副本: Subscribed ***@***.***&gt;

dongjq23, 2024-03-22:
> 请问里程计输出延时大约半分钟，这种问题应该怎么解决？ 期待您的回答
    ros::Publisher pubOdomAftMapped = nh.advertise<nav_msgs::Odometry> 
            ("/aft_mapped_to_init", 1000);
            我把发布频率从100000改到1000之后，延迟从半分钟，变成了5s左右

Xyxx3353, 2024-06-03:
请问您解决了延迟的问题了吗

INDEX108, 2024-07-12:
请问解决了么 我遇到了同样的问题

Joanna-HE, 2024-07-14:
你好～如果是高频（4khz）的里程计输出延迟问题可能是由于ros：：spin是一帧雷达进行一次的。如果需要高频非延迟里程计输出，可以修改代码的ros：：spin，试试看呢

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

INDEX108, 2024-07-14:
> 你好～如果是高频（4khz）的里程计输出延迟问题可能是由于ros：：spin是一帧雷达进行一次的。如果需要高频非延迟里程计输出，可以修改代码的ros：：spin，试试看呢 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

请问是改rosrate么 我目前Lidar是hap 10HZ

INDEX108, 2024-07-14:
> 你好～如果是高频（4khz）的里程计输出延迟问题可能是由于ros：：spin是一帧雷达进行一次的。如果需要高频非延迟里程计输出，可以修改代码的ros：：spin，试试看呢 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

我想借助point-lio优越的抗振动特性，但是里程计我只需要10HZ的 目前有明显时间延时

INDEX108, 2024-07-14:
> 你好～如果是高频（4khz）的里程计输出延迟问题可能是由于ros：：spin是一帧雷达进行一次的。如果需要高频非延迟里程计输出，可以修改代码的ros：：spin，试试看呢 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

我发现con_frame打开就可以避免时间延迟，但是里程计会到1HZ。若使用PointCloud2的激光雷达点云可以正常运行，但是使用CustomMsg格式 该情况下程序会死掉 类似bad alloc，vector length这类

Joanna-HE, 2024-07-16:
vector length这些问题大概率是和path publisher里面存储了过多的odometry有关的。

获取 Outlook for iOS<https://aka.ms/o0ukef>

INDEX108, 2024-07-16:
您好，现在我将原来的一些点降采样了，目前没有明显时延了。但是有个疑问，我看pointlio的cpu占用率好像不高，这是为什么，好像只用了两个cpu线程

Joanna-HE, 2024-07-16:
因为point-lio本质上存在一个串行执行流程，所以后续没有继续做过多线程版本开发

获取 Outlook for iOS<https://aka.ms/o0ukef>

D

INDEX108, 2024-07-16:
> 因为point-lio本质上存在一个串行执行流程，所以后续没有继续做过多线程版本开发 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)
> ________________________________ 发件人: INDEX108 ***@***.***> 发送时间: Tuesday, July 16, 2024 6:00:32 PM 收件人: [...]

如果想要改多线程的话 比较困难嘛？我不太清楚。想请教一下，主要是哪部分需要改并行多线程呢？

Ovalene2333, 2025-05-16:
> 请问里程计输出延时大约半分钟，这种问题应该怎么解决？ 期待您的回答

我也碰到了这个问题，解决方法要么是调大filter_size_surf和filter_size_scan，要么调小ivox_grid_resolution，尤其是调小ivox_grid_resolution到0.2(也就是parameters.cpp所设定的默认值)能有非常显著的改善。