hku-mars/Point-LIO #80: 关于重力矫正的问题

LittleDang, 2024-04-12:
你好，非常感谢您的工作。在实际跑起来的时候，我有一些关于重力矫正的疑问，希望能得到您的回复，十分感谢。
1. 在看代码的时候gravity是采用的vect3而不是S2，这是为什么呢？
2. 我在使用LIVOX-MID360运行（安装在车上）的时候（开启了重力对齐），发现z轴有往上翘的感觉，特别是在远离原点的地方更加明显，我监控了全程的的重力向量输出，发现它始终是接近0,0,-9.81，这是否表明在状态估计的过程中，全局坐标系下的重力始终是没有发生明显的角度变化？由于重力加速度的存在，我觉得重力应该始终不会在roll和pitch的纬度上产生累积误差才对，但是我观察到的结果似乎也会产生累积误差，一个证明的方式是我让车移动到原理原点的地方启动，在那里静止的时候直接通过IMU的重力加速度计算出来的重力方向和算法在此位置（在比较远的地方开始建图，然后跑到这个位置时）估计出来的重力方向有较大的差距（大概有十几度）。
这是我建出来的图的正视视角
[image]
[这是我使用的数据](https://drive.google.com/file/d/18KZiN79a4PliYQq8MlQCUYSw5SUe6v5m/view?usp=drive_link)

dongjq23, 2024-05-27:
> 你好，非常感谢您的工作。 在实际跑起来的时候，我有一些关于重力矫正的疑问，希望能得到您的回复，十分感谢。
> 
> 1. 在看代码的时候gravity是采用的vect3而不是S2，这是为什么呢？
> 2. 我在使用LIVOX-MID360运行（安装在车上）的时候（开启了重力对齐），发现z轴有往上翘的感觉，特别是在远离原点的地方更加明显，我监控了全程的的重力向量输出，发现它始终是接近0,0,-9.81，这是否表明在状态估计的过程中，全局坐标系下的重力始终是没有发生明显的角度变化？   由于重力加速度的存在，我觉得重力应该始终不会在roll和pitch的纬度上产生累积误差才对，但是我观察到的结果似乎也会产生累积误差，一个证明的方式是我让车移动到原理原点的地方启动，在那里静止的时候直接通过IMU的重力加速度计算出来的重力方向和算法在此位置（在比较远的地方开始建图，然后跑到这个位置时）估计出来的重力方向有较大的差距（大概有十几度），请问这是我的理解在哪个环节出错了？
> 
> **PS：我用的方式是OUTPUT模式** 这是我建出来的图的正视视角 ![2024-04-12_10-30](https://private-user-images.githubusercontent.com/30885867/321838493-86ae013c-b855-4870-b66b-1ea570150c5f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3ODUwMDMsIm5iZiI6MTcxNjc4NDcwMywicGF0aCI6Ii8zMDg4NTg2Ny8zMjE4Mzg0OTMtODZhZTAxM2MtYjg1NS00ODcwLWI2NmItMWVhNTcwMTUwYzVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI3VDA0MzgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwNDc1OWY3MzY2NWQxMzYzZWJhYWI5MTI5Y2YxZjA0ZWU3MDAxMzdjMGRkMDhkMWQyNzAxMDJmZDYyNTZmYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OvqbVUfqvv5Oxo6b8lnlrGkc8zSJ9IpWWpyccx_Ai7A) [这是我使用的数据](https://drive.google.com/file/d/18KZiN79a4PliYQq8MlQCUYSw5SUe6v5m/view?usp=drive_link)

请问您思考出原因了吗？我也有相同的问题。

Joanna-HE, 2024-05-31:
你好，感谢您的关注和提问。

  1.  point-lio目前采用的在线估计重力的方式，是不会对重力有明显的改变的。S2对重力建模会更精确，但是因为算法对重力改变量较小，所以用vect3也可以近似了。如果想进一步研究对重力估计的话，可以尝试增加init_cov中重力加速度的初始方差值。
  2.
地面上翘的过程不一定会影响重力估计。地面上翘还是因为雷达的点面匹配在测量噪声等方面建模不准确，可以影响其他状态量，比如加速度acc和姿态等，不一定会反映在重力估计上。

期待你们对重力加速度的在线估计有更新一步的进展！

________________________________
寄件者: Mandy ***@***.***>
寄件日期: 2024年5月27日 下午 12:47
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: Re: [hku-mars/Point-LIO] 关于重力矫正的问题 (Issue #80)

你好，非常感谢您的工作。 在实际跑起来的时候，我有一些关于重力矫正的疑问，希望能得到您的回复，十分感谢。

  1.  在看代码的时候gravity是采用的vect3而不是S2，这是为什么呢？
  2.  我在使用LIVOX-MID360运行（安装在车上）的时候（开启了重力对齐），发现z轴有往上翘的感觉，特别是在远离原点的地方更加明显，我监控了全程的的重力向量输出，发现它始终是接近0,0,-9.81，这是否表明在状态估计的过程中，全局坐标系下的重力始终是没有发生明显的角度变化？ 由于重力加速度的存在，我觉得重力应该始终不会在roll和pitch的纬度上产生累积误差才对，但是我观察到的结果似乎也会产生累积误差，一个证明的方式是我让车移动到原理原点的地方启动，在那里静止的时候直接通过IMU的重力加速度计算出来的重力方向和算法在此位置（在比较远的地方开始建图，然后跑到这个位置时）估计出来的重力方向有较大的差距（大概有十几度），请问这是我的理解在哪个环节出错了？

PS：我用的方式是OUTPUT模式 这是我建出来的图的正视视角 [2024-04-12_10-30] <https://private-user-images.githubusercontent.com/30885867/321838493-86ae013c-b855-4870-b66b-1ea570150c5f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3ODUwMDMsIm5iZiI6MTcxNjc4NDcwMywicGF0aCI6Ii8zMDg4NTg2Ny8zMjE4Mzg0OTMtODZhZTAxM2MtYjg1NS00ODcwLWI2NmItMWVhNTcwMTUwYzVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI3VDA0MzgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwNDc1OWY3MzY2NWQxMzYzZWJhYWI5MTI5Y2YxZjA0ZWU3MDAxMzdjMGRkMDhkMWQyNzAxMDJmZDYyNTZmYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OvqbVUfqvv5Oxo6b8lnlrGkc8zSJ9IpWWpyccx_Ai7A>  这是我使用的数据<https://drive.google.com/file/d/18KZiN79a4PliYQq8MlQCUYSw5SUe6v5m/view?usp=drive_link>

请问您思考出原因了吗？我也有相同的问题。

LittleDang, 2024-05-31:
感谢您的回复，我现在也怀疑是一些匹配方面的噪音或者外参数不准等原因会导致的无法消除的累积误差。

closed 2024-05-31 by LittleDang

moonsheep99, 2025-01-14:
您好，我也遇到了类似的问题，请问一下，我走了一个大一点的回环，从起点出发再回到终点，整体定位和建图效果都很好，终点处的z轴值也与初始z轴值一致，但是过程中就不一致了，越远离原点的时候，越不在z=0上（也测试了FAST-LIO2，是始终在z=0附近的），当返回时离原点越近时，又正常。我不清楚是重力向量的问题还是什么问题，我打印了重力向量，当在原点附近时，重力向量z接近9.8，当离原点最远时，z基本是9.5、9.6这样。非常希望能得到一些帮助，感谢！

moonsheep99, 2025-01-14:
整体建图效果是能看出来比FAST-LIO好的，但是z轴高度并不总在0附近，我的雷达是mid360，其z轴并不是垂直于地面，我也尝试过将初始旋转矩阵设定为单位阵，也是有类似的问题

Joanna-HE, 2025-01-14:
是猜测是重力估计不准导致的嘛？

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

LittleDang, 2025-01-14:
我感觉是因为匹配误差或者雷达本身的误差然后形成的地图带有一些误差（例如估计出来的角度误差很小，在原点附近造成的匹配误差比较小，而远离原点的位置就可能导致匹配误差越来越大），而地图本身只会增删，而不会去优化这些点云，导致进一步的匹配误差，误差会越来越大，最后这些误差表现在了状态估计的各个量上（包括重力估计）

moonsheep99, 2025-01-14:
真的非常感谢各位的回复！我猜测有两种1）是重力估计不准导致的；2）是世界坐标系定义导致的，但是我没进一步分析出来。大体情况是这样的：
（红色是我将rviz的fixed frame延长了，应该是世界系）
[image]
这是从起点出发，然后在走廊里远离原点，当绕了一打圈后再返回原点，在原点附近其实z轴高度基本一致，来回两次建图也是一致的（FAST-LIO就有些地图重影），但是在远离原点时，机体的z轴就不在0附近了。
但是起点和终点的一致以及地图的一致，让我觉得可能不是点云误差的问题？（这点我不确定，还望指点哈）
这是返程是的图片：
[image]

我看了下point-lio第一帧的处理，为了验证是不是坐标系定义的问题，我在程序中计算了前几秒的重力，然后把这个重力给到参数文件中的gravity: [0.162735, 0.0420642,   9.80756]
[image]
但是也是类似的现象。各位能给一些帮助或是调试的建议吗，感谢感谢！

moonsheep99, 2025-01-14:
我的mid360雷达的z轴是朝向地面的，但不是垂直于地面的，稍微有一点点斜度。如果需要的话，我将尽快将数据上传，麻烦啦！

moonsheep99, 2025-01-14:
我使用的是mid360，z轴朝向地面，但不垂直于地面，有一点点的倾斜。这是我的数据哈：[https://drive.google.com/file/d/1ihjXUJNTuK_UtYIxz2M6lFaqFvbHZ0nk/view?usp=drive_link]

moonsheep99, 2025-01-14:
我使用的是mid360，z轴朝向地面，但不垂直于地面，有一点点的倾斜。[这是我的数据哈](https://drive.google.com/file/d/1ihjXUJNTuK_UtYIxz2M6lFaqFvbHZ0nk/view?usp=drive_link)

moonsheep99, 2025-01-14:
问题大该清楚了，就像楼上同学说的，确实是激光点的误差引起的，我把点云噪声调大了一些，虽然问题没完全解决，但是不那么漂了，改善了好多，非常感谢各位的回复和解答！

LittleDang, 2025-01-15:
这样子建立出来的图应该重影会比较严重一点吧？

moonsheep99, 2025-01-15:
xy方向上其实效果很好，z轴上确实重影一些