hku-mars/Point-LIO #85: vel_cov这个参数我百思不得其解啊

liaraaaaaaa, 2024-05-21:


liaraaaaaaa, 2024-05-21:
> 可以理解成加速度的估计误差
> […](#)

感谢您的回复！但是如果在速度部分考虑加速度的估计误差的话，为什么没有在旋转部分考虑角速度的估计误差呢？这是出于什么考虑呢

Joanna-HE, 2024-05-21:
因为直接观测是p和R，a是p的二阶导，而omg是R的一阶导。
________________________________
寄件者: 许益豪@机器人组 ***@***.***>
寄件日期: 2024年5月21日 下午 05:44
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] vel_cov参数问题 (Issue #85)

可以理解成加速度的估计误差
…
________________________________ 寄件者: 许益豪@机器人组 @.> 寄件日期: 2024年5月21日 上午 10:22 收件者: hku-mars/Point-LIO @.> 副本: Subscribed @.> 主旨: [hku-mars/Point-LIO] vel_cov这个参数我百思不得其解啊 (Issue #85<https://github.com/hku-mars/Point-LIO/issues/85>) D

感谢您的回复！但是如果在速度部分考虑加速度的估计误差的话，为什么没有在旋转部分考虑角速度的估计误差呢？这是出于什么考虑呢