hku-mars/Point-LIO #87: IMU initialization requires static or dynamic start

CharlieV5, 2024-05-26:
Hi, Dr. He!
I notice that the latest commits are about the bug for unstationary start. I would like to ask whether IMU initialization requires the device to be in a static state at startup or if it can be in motion as well?  I looked at your related code, and I guess the code probably only handles static situations. I've tested some data with a startup in motion and the initial map may be not high-precision with "ghosting effect".

Joanna-HE, 2024-05-31:
Hi, thanks for your question!

Yes, the point-lio relies on the initial IMU measurements to estimate the initial gravity value in the starting IMU coordinate. Therefore, in general, it is required to have a static start situation. However, if the start velocity and acceleration is not large, Point-lio can still work with somehow degraded performance at the start.
________________________________
寄件者: Charlie Wang ***@***.***>
寄件日期: 2024年5月26日 下午 11:04
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] IMU initialization requires static or dynamic start (Issue #87)

Hi, Dr. He!
I notice that the latest commits are about the bug for unstationary start. I would like to ask whether IMU initialization requires the device to be in a static state at startup or if it can be in motion as well? I looked at your related code, and I guess the code probably only handles static situations. I've tested some data with a startup in motion and the initial map may be not high-precision with "ghosting effect".

CharlieV5, 2024-05-31:
Thanks for your reply!

closed 2024-05-31 by CharlieV5