hku-mars/Point-LIO #88: 点云模糊问题

seungjuuuuuu, 2024-06-04:
你好！感谢你优秀的开源代码，我在运行自己录制的包的时候，发现建图出的点云非常模糊，请问这是什么情况？（数据包是在纹理特征比较少的矿区录制的）
[image]

Joanna-HE, 2024-06-04:
看起来是位姿飞了。可以查看下加速度计的单位和采集点云的时间戳的单位哈。
________________________________
寄件者: Shengz Li ***@***.***>
寄件日期: 2024年6月4日 上午 11:05
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 点云模糊问题 (Issue #88)

This is an external email.

你好！感谢你优秀的开源代码，我在运行自己录制的包的时候，发现建图出的点云非常模糊，请问这是什么情况？（数据包是在纹理特征比较少的矿区录制的）
download.png (view on web)<https://github.com/hku-mars/Point-LIO/assets/129495802/30d3dc15-fdcc-4a20-a7a6-6d846861dbb0>

seungjuuuuuu, 2024-06-07:
贺博士你好，关于你新提交的branch voxel map point-lio，它对抖动得处理更加鲁棒了，请问你是否只对ikdtree替换为了voxel进行了操作呢？