hku-mars/Point-LIO #91: 关于高频率输出里程计后，状态变量的速度估计输出会有很多噪声，不利于控制。

19855179496, 2024-07-11:
[image]
[image]
[image]

我们开启了高频率的输出，然后将kf_output的速度作为输出导出来，作为控制器的输入。但发现无人机总是失控，最后利用软件工具发现输出的线速度三轴分量都有着比较明显的噪声。同时我们还与别的算法进行比较，发现确实有着明显的区别（前两张图）。请问作者大大，这个问题怎么解决。（tip：第三张图是位置的输出效果，还是很平滑正常的）

19855179496, 2024-07-11:
还有一个问题就是关于欧拉角初始值的问题，我们发现在一开始point_lio输出的欧拉角不是0（按理说应该为0比较合理），这个现象是什么导致的呢

19855179496, 2024-07-11:
[image]

ysingfun, 2024-08-15:
我也感觉有类似的问题，每个点的对应位姿高频噪声很大！我使用每个点云点对应位姿，将帧内每个点云点对齐到该帧同一个位姿上（帧内最后一个点云点的位姿）时，建图结果相对于作者的发散很多，漂移很大，但是还可以基本上建完图。
我后面使用帧内点云位姿的均值，将所有点云都对其该均值上，发散程度改善一点，但是还是比作者的差好多。
之后我对所有点云点的进行滑动窗均值滤波，再对齐到帧内所有点位姿均值上，并没有任何改善效果。
我猜测，作者的建图一致性比我对齐的效果好很多，可能时因为随机一致性的原由。虽然每个点的高频噪声很大，不确定性很高，但是服从类似高斯分布的，很好的分布在真实值附近，整体上建图一致性效果非常好！感慨作者水平非常厉害，算法非常高明！
但是我也很好奇，怎么估计过程中去除这种高频噪声？请问有什么好的思路？