hku-mars/Point-LIO #92: 地图z处理问题

dongjq23, 2024-07-13:
感谢作者优秀的开源！

dongjq23, 2024-07-13:
### 感谢作者优秀的开源！
**因为我应用的场景比较特殊（船舱环境），项目上有一些对地图处理的需求。但是我对于point-lio的理解不够深入，不知道需要怎么修改才能满足我的要求。**

我的雷达是mid360，采回来的一些数据包在这里：链接: https://pan.baidu.com/s/1PmE2aJVYI2zvz-Do4kVHZw?pwd=jlu6 提取码: jlu6 
播放数据集后，启动 mapping\_mid360.launch 文件开始建图。（没有对于任何源码进行修改）

*   **mid360.yaml文件如下：(因为点云比较多，有做了下采样的预处理，不过yaml里订阅的是原点云)**

```yaml
common:
    lid_topic:  "/livox/lidar"  #"/filtered_cloud/intensity" 
    imu_topic:  "/livox/imu_192_168_1_151" 

    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency
    # time_lag_imu_to_lidar: 0.3 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value
    time_lag_imu_to_lidar: 0.001792 #-0.009263
preprocess:
    lidar_type: 2 
    scan_line: 6
    timestamp_unit: 1           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1.0 #0.5 

mapping:
    imu_en: true
    start_in_aggressive_motion: false # if true, a preknown gravity should be provided in following gravity_init
    extrinsic_est_en: true # for aggressive motion, set this variable false
    imu_time_inte: 0.005 # = 1 / frequency of IMU
    # satu_acc: 3.0 # the saturation value of IMU's acceleration. not related to the units
    satu_acc: 4.0
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 1.0 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.1 #0.001 # 0.001; 0.01
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.0001 
    b_gyr_cov: 0.0001 
    imu_meas_acc_cov: 0.1 #0.1 # 0.1
    imu_meas_omg_cov: 0.1 #0.01 # 0.1
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 180 #360 
    det_range: 50.0
    #True表示将世界框架的z轴与重力方向对齐，重力方向应在下面指定
    gravity_align: false # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity:  [0.0, 0.0, -9.800291]  # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.800291] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    # extrinsic_T: [ 0.04165, 0.02326, -0.0284 ]
    extrinsic_T: [ 0.011, 0.02329, -0.04412 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ]

odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

*   **mapping\_mid360.launch文件如下：**

```bash
<launch>
<!-- Launch file for Livox AVIA LiDAR -->

	<arg name="rviz" default="true" />

	<node pkg="point_lio" type="pointlio_mapping" name="laserMapping" output="screen"> 
	<rosparam command="load" file="$(find point_lio)/config/mid360.yaml" />
	<param name="use_imu_as_input" type="bool" value="0"/> <!--change to 1 to use IMU as input of Point-LIO-->
	<param name="prop_at_freq_of_imu" type="bool" value="1"/>
	<param name="check_satu" type="bool" value="1"/>
	<param name="init_map_size" type="int" value="10"/>
	<param name="point_filter_num" type="int" value="1"/> <!--4, 3--> 
	<param name="space_down_sample" type="bool" value="1" />  
	<param name="filter_size_surf" type="double" value="0.3" /> <!--0.5, 0.3, 0.2, 0.15, 0.1--> 
	<param name="filter_size_map" type="double" value="0.2" /> <!--0.5, 0.3, 0.15, 0.1-->
	<param name="cube_side_length" type="double" value="2000" /> <!--1000-->
	<param name="runtime_pos_log_enable" type="bool" value="0" /> <!--1-->
	</node>
	<group if="$(arg rviz)">
	<node launch-prefix="nice" pkg="rviz" type="rviz" name="rviz" args="-d $(find point_lio)/rviz_cfg/loam_livox.rviz" />
	</group>

	launch-prefix="gdb -ex run --args"

</launch>
```

### 问题一：地图绕z轴的偏航角度偏差

建图过程中，效果如下：
[image]
因为建图初始帧，**imu坐标系**与 **船舱两垂直侧壁和底面 构成的坐标系**不是平行的。所以船舱结构俯视投影与grid map存在偏航角偏差。
**我想消除上述的偏差，保持建图过程中，船舱俯视投影与grid map能够水平对齐。我应该如何去做呢？**
我能想到的时，在建图初始时刻，保持雷达不动，**利用实时的点云数据感知拟合船舱侧壁平面**。选定**船舱一角以及垂直侧面和舱底面作为坐标系**。 计算出IMU坐标系相对于选定的船体坐标系的x y方向偏差，以及偏航角度偏差。
**在计算出上述的偏差值之后，我应该将偏差值，写入point-lio源码的哪个位置呢？**
下面的建图结束后得到的PCD：
[image]

### 问题二：生成的地图绕x和绕y方向存在3度以内的偏差
问题也是出在**初始IMU**上，初始点云转换成PCD后查看，就有倾斜的状态。但是无论我在yaml中 **设置IMU是否与重力对齐**（gravity\_align: false 或者 true），都会有倾斜的情况出现。非静止起动情况下，IMU偏转更大，地图生成后偏差也更大。
这是bag数据初始7帧点云叠加的侧视图和正视图：
[image]
[image]

针对上述的两个问题，您可以帮忙分析一下问题的本质原因和我想的是一样的吗？可以提供一些解决方案嘛？

Joanna-HE, 2024-07-14:
你好，livox mid360和livox avia是用一样的雷达type，也就是lidar_type要改为1

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

<launch>

2175mapping.png (view on web)<https://github.com/user-attachments/assets/3d7a1286-cb2c-4087-8d41-6fa5946b3032>
2175_scans.png (view on web)<https://github.com/user-attachments/assets/12991c92-87f6-4011-8dcd-ad1d10f1e3f3>

2175_first_7frames_left.png (view on web)<https://github.com/user-attachments/assets/f07ce207-5789-431b-a89f-e998e3063818>
2175_first_7frames_front.png (view on web)<https://github.com/user-attachments/assets/9d057404-f618-46ca-a382-4d77a9b8d8dc>

dongjq23, 2024-07-29:
> 你好，livox mid360和livox avia是用一样的雷达type，也就是lidar_type要改为1 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

谢谢作者！我把lidar_type改为 1 了，后续接收点云会继续使用livox_ros_driver2/CustomMsg格式的。我也通过向别人请教解决了地图倾斜的问题。再次感谢作者优秀的开源工作！！

closed 2024-07-29 by dongjq23

channing-prince, 2025-11-11:
您好@dongjq23 ，我也遇到了相同问题，我尝试了用飞控IMU的重力方向去给点云的世界坐标做变换，还是有较小偏差，这里问题是激光雷达本身在无人机上安装的偏差，还是其他原因导致的呢，我应该怎么确保点云是与重力方向对齐的