hku-mars/Point-LIO #96: Mapping Problem on MCD Dataset

chengwei0427, 2024-08-29:
Hi, Dear Author,
Thanks for your nice work.
I have recently been testing your open-source algorithm and noticed some drift when working with MCD data. Could you suggest which parameters I could adjust to optimize the results for this dataset？
[image]
[image]
[image]
Thanks.

Joanna-HE, 2024-08-29:
Try to set acc_norm = 9.805, satu_acc = 30.0
And set gravity=[0.0, 0.0, -9.805]

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

2024-08-29.11-56-13.png (view on web)<https://github.com/user-attachments/assets/0ffa6529-2441-45c3-9afe-bfdd74c94d66>
2024-08-29.11-56-45.png (view on web)<https://github.com/user-attachments/assets/2d6ce8fb-7ea3-4b32-a91d-62ae020a079d>
image.png (view on web)<https://github.com/user-attachments/assets/522bcfeb-f4ec-4ec5-b738-da14b3c66814>

Joanna-HE, 2024-08-29:
What’s more, use point-lip-grid-map branch to have a more stable performance

获取 Outlook for iOS<https://aka.ms/o0ukef>

closed 2024-09-03 by chengwei0427