hku-mars/Point-LIO #97: 关于状态更新后的协方差矩阵

LittleDang, 2024-08-30:
作者你好，我在观看代码和论文的时候发现代码里面在状态更新后协方差矩阵并没有像论文一样进行最后一步（公式24）的变换。
请问这是为什么呢？

这是代码里的状态更新
[image]
这是论文里对应的部分
[image]

Joanna-HE, 2024-08-30:
cov reset是做了一次误差空间重投影。如果residual非常小的话，这一步收益不大。

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

update.png (view on web)<https://github.com/user-attachments/assets/5d23f7df-9bbe-4121-8d82-e17c983fef6c>
J.png (view on web)<https://github.com/user-attachments/assets/8a0d5f7d-ee56-49e2-8e38-eb8e8b876577>

LittleDang, 2024-08-30:
懂啦，感谢回复

closed 2024-08-30 by LittleDang