hku-mars/Point-LIO #102: Fast-Lio2 Performs Better

celikemir, 2024-10-11:
Hi,

I have several bag files recorded at different altitudes with a UAV, such as 50m, 100m, etc. The LiDAR is positioned at nadir, pointing downward. There is significant odometry degradation, particularly in some areas. I also tested the same bag files with FAST-LIO2, and its performance was better and closer to the ground truth. My question is: do you have any suggestions for adjusting parameters for outdoor use and fixing these degradation points? (I've already adjusted filter_size_surf, etc.) The odometry completely diverges in the opposite direction and fails to recover.

sharknow, 2024-10-11:
> Hi,
>
> I have several bag files recorded at different altitudes with a UAV, such as 50m, 100m, etc. The LiDAR is positioned at nadir, pointing downward. There is significant odometry degradation, [...]

Can you provide a picture with the results? Thank you!

celikemir, 2024-10-11:
Here are the results below, where you can easily distinguish which one diverges from the ground truth.

[image]

[image]

[image]

sharknow, 2024-10-11:
1987年9月14日21时07分 中国第一封电子邮件 从北京发往德国 "越过长城,走向世界" QQ邮箱,常联系!您好，这是来自QQ邮箱的自动回复邮件。