hku-mars/Point-LIO #105: 关于F矩阵和P矩阵

sharknow, 2024-10-28:
请问为什么代码里面的F矩阵和论文对不上呢？
用的是EKF还是ESKF？
P阵是状态的协方差矩阵还是误差状态的协方差矩阵？
[image]

Joanna-HE, 2024-10-28:
论文的F矩阵是x-error_{k+1}对x-error_{k}的导数结果。代码里面的df_dx是f函数对x-error_{k}的导数，可以计算出F矩阵的一部分。其中f的定义为x_{k+1}=x_{k}+f(x_{k}). 论文中F的表达式可以在esekfom.hpp的predict函数中找到，表达式为多个部分的加和。
________________________________
寄件者: wanghe ***@***.***>
寄件日期: 2024年10月28日 下午 03:44
收件者: hku-mars/Point-LIO ***@***.***>
副本: Subscribed ***@***.***>
主旨: [hku-mars/Point-LIO] 关于F矩阵和P矩阵 (Issue #105)

This is an external email.

请问为什么代码里面的F矩阵和论文对不上呢？
用的是EKF还是ESKF？
P阵是状态的协方差矩阵还是误差状态的协方差矩阵？
image.png (view on web)<https://github.com/user-attachments/assets/0729d60b-5dc8-4e3a-8465-104cc692f4c1>

bingquanLi, 2024-10-28:
if(!prop_at_freq_of_imu)//原始为1 
{   
    double dt_cov = time_current - time_update_last;
    if (dt_cov > 0.0)
    {    
        kf_input.predict(dt_cov, Q_input, input_in, false, true); 
        time_update_last = time_current; 
    }
}
原始配置为1，好像没有调过kf_input.predict(dt_cov, Q_input, input_in, false, true); 想问一下是不需要方差传播吗？

sharknow, 2024-10-29:
> 论文的F矩阵是x-error_{k+1}对x-error_{k}的导数结果。代码里面的df_dx是f函数对x-error_{k}的导数，可以计算出F矩阵的一部分。其中f的定义为x_{k+1}=x_{k}+f(x_{k}). 论文中F的表达式可以在esekfom.hpp的predict函数中找到，表达式为多个部分的加和。
> […](#)

ok，谢谢您！

sharknow, 2024-10-29:
> if(!prop_at_freq_of_imu)//原始为1 { double dt_cov = time_current - time_update_last; if (dt_cov > 0.0) { kf_input.predict(dt_cov, Q_input, input_in, false, true); time_update_last = time_current; } } 原始配置为1，好像没有调过kf_input.predict(dt_cov, Q_input, input_in, false, true); 想问一下是不需要方差传播吗？

需要方差传播，如果不调kf_input，那就是调了kf_output。kf_output里面会进行cov的传播。

sharknow, 2024-10-30:
> 论文的F矩阵是x-error_{k+1}对x-error_{k}的导数结果。代码里面的df_dx是f函数对x-error_{k}的导数，可以计算出F矩阵的一部分。其中f的定义为x_{k+1}=x_{k}+f(x_{k}). 论文中F的表达式可以在esekfom.hpp的predict函数中找到，表达式为多个部分的加和。
> […](#)

请问F_11为什么没在代码里体现，没找到这一项的赋值？谢谢！
[image]