hku-mars/Point-LIO #112: The node doesn't save the map

Andy-Leo10, 2025-02-18:
Hi, great work!

I have a problem, the map is not being saved. I attach the content of my `ouster64.yaml` maybe there is something wrong

```
common:
    lid_topic:  "/ouster/points"
    imu_topic:  "/ouster/imu"
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency 
    time_diff_lidar_to_imu: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)

preprocess:
    lidar_type: 3 # 2 #velodyne                # 1 Livox Avia LiDAR
    scan_line: 32 # 32 #velodyne 6 avia
    timestamp_unit: 3           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 4.0

mapping:
    imu_en: true
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.01 # = 1 / frequency of IMU
    lidar_time_inte: 0.1
    satu_acc: 30.0 # the saturation value of IMU's acceleration. not related to the units
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.81 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.01 # 0.001
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.0001 
    b_gyr_cov: 0.0001 
    imu_meas_acc_cov: 0.1 #0.1 # 2
    imu_meas_omg_cov: 0.1 #0.1 # 2 
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.1 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    ivox_grid_resolution: 2.0
    gravity: [0.0, 0.0, -9.810] # preknown gravity, use when imu_en is false or start from a non-stationary state
    gravity_init: [0.0, 0.0, -9.810] # preknown gravity in the initial IMU frame for unstationary start or in the initial LiDAR frame for using without IMU
    extrinsic_T: [0.0, 0.0, 0.0]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ]

odometry: 
    publish_odometry_without_downsample: false

publish:
    path_en: true                 # false: close the path output
    scan_publish_en: true         # false: close all the point cloud output
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: 1
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

Joanna-HE, 2025-02-19:
please make sure you have the local file folder with the right name

获取 Outlook for iOS<https://aka.ms/o0ukef>

This is an external email.

Andy-Leo10, 2025-02-19:
Yes, I created inside the package a folder named "PCD" with a file named "scans.pcd". But when my rosbag ends, and then I press crt+c the map is not being saved. Am I doing something wrong?

Joanna-HE, 2025-02-20:
please delete the '&& pcd_save_interval > 0' at the line 212 of the file laserMapping.cpp and create a folder named "PCD" at the same level as Log folder.
________________________________
寄件者: Andrés Leonardo ***@***.***>
寄件日期: 2025年2月20日 上午 12:23
收件者: hku-mars/Point-LIO ***@***.***>
副本: ***@***.*** ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/Point-LIO] The node doesn't save the map (Issue #112)

This is an external email.

[Andy-Leo10]Andy-Leo10 left a comment (hku-mars/Point-LIO#112)<https://github.com/hku-mars/Point-LIO/issues/112#issuecomment-2669139652>

Yes, I created inside the package a folder named "PCD" with a file named "scans.pcd". But when my rosbag ends, and then I press crt+c the map is not being saved. Am I doing something wrong?

Andy-Leo10, 2025-02-20:
Thank you so much! that solved the problem and I obtained the map

closed 2025-02-20 by Andy-Leo10