hku-mars/Point-LIO #115: Config parameters when tilting lidar

migsdigs, 2025-06-26:
Hi there, thank you very much for your great work!

I am using the mid-360 lidar with point-lio. I would like to clarify some questions I have about the configuration parameters when I would like to tilt the lidar instead of using it flat mounted. When I am using it flat mounted, I am using the following config params (note: this is in simulation, so the imu sits in the same frame as the lidar source):
```
common:
    lid_topic:  "/livox/lidar" 
    imu_topic:  "/livox/imu"
#    imu_topic: "/uav1/hw_api/imu"
    con_frame: false # true: if you need to combine several LiDAR frames into one
    con_frame_num: 1 # the number of frames combined
    cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
    cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency
    time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                               # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value

preprocess:
    lidar_type: 1 
    scan_line: 6
    timestamp_unit: 1           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1.0                  # also from https://github.com/hku-mars/Point-LIO/issues/24, sometimes 0.5?

mapping:
    imu_en: true
    extrinsic_est_en: false # for aggressive motion, set this variable false
    imu_time_inte: 0.01 # = 1 / frequency of IMU
    satu_acc: 4.0 # the saturation value of IMU's acceleration. not related to the units # TODO: should be 4.0 not 30.0?
    satu_gyro: 35 # the saturation value of IMU's angular velocity. not related to the units
    acc_norm: 9.81 # 1.0 for g as unit, 9.81 for m/s^2 as unit of the IMU's acceleration
    lidar_meas_cov: 0.1 # 0.001; 0.01
    acc_cov_output: 500
    gyr_cov_output: 1000 
    b_acc_cov: 0.0001 
    b_gyr_cov: 0.0001 
    imu_meas_acc_cov: 0.01 #0.1 # 0.1
    imu_meas_omg_cov: 0.1 #0.01 # 0.1
    gyr_cov_input: 0.01 # for IMU as input model
    acc_cov_input: 0.1 # for IMU as input model
    plane_thr: 0.05 # 0.05, the threshold for plane criteria, the smaller, the flatter a plane
    match_s: 81
    fov_degree: 360 
    det_range: 50.0
    gravity_align: false # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # # preknown gravity in the first IMU body frame, use when imu_en is false or start from a non-stationary state
    # TODO: is this true for MID360 Livox?
    extrinsic_T: [ 0.0, 0.0, 0.0 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1 ]

#    extrinsic_T: [ 0.074, 0.0, 0.104 ]

odometry: 
    publish_odometry_without_downsample: false

publish:
    path:
      enable: true # false: close the path output
      # TODO: use these values in code
      diff_t: 0.01 # (m) publish when translation has changed by this value
      diff_R: 0.01 # (rad) publish when one of the Euler angles has changed by this value
    scan_publish_en: true       # false: close all the point cloud output   TODO: might want to enable this and disable below
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

If I want to pitch the lidar for example 25 degrees down, so that it sees more of the floor, do I need to alter any of the parameters relating to gravity i.e. 
```
    gravity_align: false # true to align the z axis of world frame with the direction of gravity, and the gravity direction should be specified below
    gravity: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] # gravity to be aligned
    gravity_init: [0.0, 0.0, -9.810] # [0.0, 9.810, 0.0] 
```

or any other parameters?

Your help would be greatly appreciated, thank you.

Kind regards

pickles976, 2025-12-11:
Did you figure this out?

migsdigs, 2025-12-14:
> Did you figure this out?

I ended up using [fast-lio](https://github.com/hku-mars/FAST_LIO) as it was more stable.