hku-mars/Point-LIO #116: 里程计话题相关问题

NarcissusEcho, 2025-06-29:
你好，我在将Point-LIO适配Mid360时发现没有输出/Odometry话题，请问是什么原因呢？

SuperLDG, 2025-07-28:
> 
> 
> 
> > > > > Hello, when I adapted Point-LIO to Mid360, I found that there was no output /Odometry topic. What is the reason?
> > > > 
> > > > 
> > > > Have you solved it? I have the same problem.
> > > 
> > > 
> > > Do you have `/odomAftMapped` topic?
> > > No
> 
> Do you have any topic publishing from Point-LIO? If not, it might be a problem with your lidar data. If you are using hardware, make sure the output data format from the livox is 1.

Other topics are released normally, and the drawing is also in progress normally.

qiurongcan, 2025-12-11:
> > > > > > 你好，当我将Point-LIO改编到Mid360时，发现没有输出/里程计主题。原因是什么？
> > > > > 
> > > > > 
> > > > > 你解决了吗？我也有同样的问题。
> > > > 
> > > > 
> > > > 你有话题吗？ 不`/odomAftMapped`
> > 
> > 
> > 你们有Point-LIO的主题出版吗？如果没有，可能是你的激光雷达数据出了问题。如果你用的是硬件设备，确保Livox的输出数据格式是1。
> 
> 其他主题正常发布，抽奖也正常进行中。
> 
> 运行“rostopic list”时，你可以看到有Odomry主题，但没有订阅主题的数据。

代码中输出里程计的话题如下：
ros::Publisher pubOdomAftMapped = nh.advertise<nav_msgs::Odometry> 
            ("/aft_mapped_to_init", 1000);
因此里程计的话题为  /aft_mapped_to_init